eclipse-upm/upm #533: Getting error (swig) after calling deviceFound() of DS18B20

nazmul21, 2017-03-20:
I'm trying to run the DS18B20 sensor example on my Siemens IoT2020. But, I got the following error (Please check the image below). I guess, there is some problem with my swig. Please suggest me the way to get rid of it. - Thanks

[image]

nazmul21, 2017-03-20:
Okay, got it. I'd not installed the swig :/ After installing it working fine :)

jontrulson, 2017-03-20:
Hi, try:

`sensor.devicesFound()`

ie: I think you forgot an 's' there.  Or you can just use the ds18b20.py python example in the UPM repository.

nazmul21, 2017-03-21:
Hey Jon,

There was no 'swig' installed, after installation it's working fine.

Thank you very much. Highly appreciated of your code inspection :)

closed 2017-03-21 by nazmul21