eclipse-upm/upm #535: UPM Invalid Argument: Invalid GPIO pin specified

TNAnouar, 2017-03-21:
hi,
I can t run the example for the sx1276 on my raspberry pi 3 with python i tried to use python3.4 i got the next error:
Traceback (most recent call last):
  File "sx1276-lora.py", line 93, in <module>
    main()
  File "sx1276-lora.py", line 30, in main
    sensor = sensorObj.SX1276()
  File "/usr/local/lib/python2.7/site-packages/upm/pyupm_sx1276.py", line 964, in __init__
    this = _pyupm_sx1276.new_SX1276(*args)
ValueError: UPM Invalid Argument: Invalid GPIO pin specified

Any one can help me,thnx

nazmul21, 2017-03-21:
Hi
I think you need to assign pins (cs, reset, dio0...5) for PRi board. sx1276-lora.py uses Arduino style pin assignment to connect with SX1276.
---Thanks

TNAnouar, 2017-03-22:
Good morning,
Thank you for your quick response .I have tried what you told to but even i had changed the constructor i only got he same issue 
Traceback (most recent call last):
  File "sx1276-lora.py", line 96, in <module>
    main()
  File "sx1276-lora.py", line 32, in main
    sensor = sensorObj.SX1276(0x12 , 1 ,  36, 29,  31, 13,  11, 22,  37 ,7)
  File "/usr/local/lib/python2.7/site-packages/upm/pyupm_sx1276.py", line 977, i                                                                                                             n __init__
    this = _pyupm_sx1276.new_SX1276(chipRev , bus, cs, resetPin, dio0, dio1, dio                                                                                                             2, dio3, dio4, dio5)
ValueError: UPM Invalid Argument: Invalid GPIO pin specified

Any other ideas. thnx

malikabhi05, 2017-03-23:
Hi @TNAnouar , you're getting this error because the pin numbers beyond 26 are not supported in mraa for the raspberry pi3 platform. pi3 support was added fairly recently in mraa. [Here](https://github.com/intel-iot-devkit/mraa/blob/master/src/arm/raspberry_pi.c) is a link to the platform file in mraa in case you're interested in taking a look.

Propanu, 2018-10-09:
Probably bad detection of pi3+ as pi1b. There's a good likelihood [this MRAA patch](https://github.com/intel-iot-devkit/mraa/pull/928) fixes this. Feel free to reopen otherwise.

closed 2018-10-09 by Propanu