eclipse-upm/upm #538: [node.js] Illegal arguments for construction of _exports_MMA7660

ckunta, 2017-03-31:
I am using the sample code for accelerometer MMA7660 and received the error.
The same code used to work in previous version, I just updated the libraries on board (Intel Edison using XDK).

MRAA Version: v1.5.1
ERROR: /home/root/.node_app_slot/accelerometer.js:31
var myDigitalAccelerometer = new digitalAccelerometer.MMA7660(
                             ^

Error: Illegal arguments for construction of _exports_MMA7660
    at Error (native)
    ....

The code is:
var digitalAccelerometer = require('jsupm_mma7660');

// Instantiate an MMA7660 on I2C bus 0
var myDigitalAccelerometer = new digitalAccelerometer.MMA7660(
					digitalAccelerometer.MMA7660_I2C_BUS, 
					digitalAccelerometer.MMA7660_DEFAULT_I2C_ADDR);

jontrulson, 2017-03-31:
That driver has changed some (been rewritten in C).  Since you are using the defaults, you do not even need to specify the  digitalAccelerometer.MMA7660_I2C_BUS, and digitalAccelerometer.MMA7660_DEFAULT_I2C_ADDR arguments.

In addition, MMA7660_I2C_BUS was changed to MMA7660_DEFAULT_I2C_BUS in the new version.

Propanu, 2017-04-05:
@ckunta, would you please confirm the suggested change from @jontrulson solves the issue?

ckunta, 2017-04-06:
Yes it does.  Apparently my earlier response didn't get posted.

Thank you Jon.  Apparently the problem was the name change.
Here are two changes that I have to made after the sample code.

      myDigitalAccelerometer = new  digitalAccelerometer.MMA7660 (digitalAccelerometer.MMA7660_DEFAULT_I2C_BUS,digitalAccelerometer.MMA7660_DEFAULT_I2C_ADDR);

      myDigitalAccelerometer.setSampleRate(digitalAccelerometer.MMA7660_AUTOSLEEP_64);

I did some look-around and noticed that this changes have been made in github (https://github.com/intel-iot-devkit/upm/blob/master/examples/javascript/mma7660.js), but somehow the sample code in XDK still have the old version (and I just verified this on the latest build 3922).

btw, I responded earlier thru email since the message saying I can do so, but apparently the interface from email didn't work/post my message. I apologize for that.

closed 2017-04-06 by ckunta