eclipse-upm/upm #552: Help to create PR to do changes in smartdrive upm sensor

neuberfran, 2017-05-16:
Hi,

I do this https://github.com/intel-iot-devkit/upm/pull/551 but I honestly do not know if I started the PR process correctly. 

How to I create a PR do to changes in files smartdrive.cxx and smartdrive.hpp of the smartdrive upm sensor?

neuberfran, 2017-05-17:
I need urgent change code of the smartdrive to use in gradle of the android things project
I'm a coauthor of this project/upm sensor

this in smartdrive UPM https://drive.google.com/file/d/0B9gPdWq4OH-XX2lJb2pvUGRrZHM/view?usp=sharing

#461

Please change https://github.com/intel-iot-devkit/upm/blob/master/src/smartdrive/smartdrive.cxx and https://github.com/intel-iot-devkit/upm/blob/master/src/smartdrive/smartdrive.hpp for this

https://github.com/neuberfran/SmartDriveCorrect/blob/master/smartdrive.cxx and this https://github.com/neuberfran/SmartDriveCorrect/blob/master/smartdrive.hpp

**I did several tests and I advise that the best thing to do is**

**I need this urgent to put upm-smartdrive-1.2.0/ in gradle**

THis is a history/reference

#509 (comment)
https://communities.intel.com/thread/106315
#461

neuberfran, 2017-05-17:
I stay trying this tutorial (from jontrulson)

clone the upm repo
create a new branch, call it something like "smartdrive_fixes" (git checkout -b smartdrive_fixes)
make whatever changes to the smartdrive src code needed to make it work
test it.
commit the changes (with your Signed-off-by) in the commit message.
push it to your repo (git push -u origin smartdrive_fixes), creating a new branch (smartdrive_fixes) in your public github repo
from that, create the PR.

**I have errors with git push -u origin smartdrive-fixes**

https://drive.google.com/file/d/0B9gPdWq4OH-XMUx1N3VFYlBIM3c/view?usp=sharing

jontrulson, 2017-05-17:
Ok, so it seems you forgot step 2 (create a new branch, call it something like "smartdrive_fixes" (git checkout -b smartdrive_fixes)).  This explains the errors you got.  You made changes to your mastr branch, committed them, then tried to push to another (non-existant) branch.

What you ended up doing instead is adding the fixes to your master branch.  You should never commit anything to the master branch - instead you should keep it synchronized with the upstream master branch.  Also, I noticed that your upm repo fork is 2 years out of date.  Creating topic branches off of this version of master will almost certainly cause problems when you try to submit a PR based off of it.

In your case, I would just check out a new version of upm, and update your master so that it matches current upstream upm master.

Once you've done that, start with step 2 again.

neuberfran, 2017-05-17:
OK. I stay try but Please Tell me if you forgot step five: add . 
I need put add . ?

Propanu, 2017-05-17:
After you modified all the files you want, use `git add -A` and it will stage all the changes for you. Then commit your changes using something like `git commit -s -m "smartdrive: fixed x and y"`

neuberfran, 2017-05-17:
issues now :

[image]

Propanu, 2017-05-17:
You can't push directly to the intel-iot-devkit org. You'll have to work with your own fork of the UPM repository: `git remote set-url origin https://github.com/neuberfran/upm.git`. Once you update your copy, you'll be able to open a new PR against `intel-iot-devkit/upm` from your new branch.

neuberfran, 2017-05-17:
[image]

But my smartdrive files in my github fork upm repository not change nothing yet:
https://github.com/neuberfran/upm/blob/master/src/smartdrive/smartdrive.cxx

Propanu, 2017-05-17:
They would be on your new `smartdrive_fixes` branch: https://github.com/neuberfran/upm/blob/smartdrive_fixes/src/smartdrive/.

neuberfran, 2017-05-17:
I do
https://github.com/intel-iot-devkit/upm/pull/554

neuberfran, 2017-05-18:
Please, tell me why files smartdrive.cxx and smartdrive.hpp in upm sensor not have new changes yet

neuberfran, 2017-05-18:
How to I change commit message from smartdrive: fixed x and y  to upgrade smartdrive.cxx and smartdrive.hpp?

Propanu, 2017-05-18:
Run `git commit --amend` and it will open the editor to change the commit message. You can also add any code file changes with `git commit -a --amend`. Once done, do a forced push `git push -f` from the smartdrive_fixes branch.

neuberfran, 2017-05-19:
I do this 

Commit message needs to be updated, I used "x" and "y" as placeholders for actual description of changes.
Author should not be removed and replaced, it's fine if you want to add yourself to the list though.
There are a couple of instances where you replace a symbol with the exact same numerical value in the .hpp file, e.g. CMD_S == 0x53. This is not needed.
In some places you use a hard-coded '5' when calling writeArray(), but the array is larger than 5 bytes, throwing away the rest of the command.
You shouldn't make repeated calls to i2c.address since it's saved in the context (see #547).
Formatting/indentation should match existing code (you have extra spaces, tabs, etc.)

But i cant create a new PR

[image]

Propanu, 2017-05-19:
You need to work on, and compare the `smartdrive_fixes` branch. Technically, you don't have to touch the PR after you open it, just update that branch and Github will update the PR automatically for you.

neuberfran, 2017-05-19:
new error:

[image]

neuberfran, 2017-05-29:
whats is conflict?
https://github.com/intel-iot-devkit/upm/pull/557

neuberfran, 2017-06-16:
Why the smartdrive.cxx and smartdrive.hpp files have not yet been changed?

https://github.com/intel-iot-devkit/upm/pull/565

neuberfran, 2018-01-30:
Driver funcionou no Android Things DP3 / intel edison / jcenter / gradle