eclipse-upm/upm #553: replace @ref links (and #labels) in documentation

tony--, 2017-05-17:
I was reading through the documentation on github and was wondering what the @ref entries sprinkled throughout were for (e.g., [on the porting page](https://github.com/intel-iot-devkit/upm/blob/master/docs/porting.md#adding-a-new-module-to-upm) it says "(see @ref naming)").  Obviously this is for linking; in [the generated version of the docs published on iotdk.intel.com](https://iotdk.intel.com/docs/master/upm/porting.html) the "@ref ..." has been replaced with a link.

After some searching I found the explanation in [the doxygen documentation](http://www.stack.nl/~dimitri/doxygen/manual/markdown.html#markdown_dox).  This looks good on iotdk.intel.com but it doesn't look good on github.com.  

It would be nice if there was a way to have working links on github.com without losing the linking in the generated docs.  According to [the doxygen documentation](http://www.stack.nl/~dimitri/doxygen/manual/markdown.html#md_links), inline links are supported.  Would it work to replace the @ref entries with inline relative links (similar to the main [README](https://github.com/intel-iot-devkit/upm/blob/master/README.md)) in order to have working links on both github.com and iotdk.intel.com?  This change would also allow the removal of the doxygen labels, which are stripped from the generated docs but look weird when reading on github.com.

Propanu, 2017-05-19:
Hi @tony--, I've looked into this a long time ago and there was no easy way around it. Doxygen has markdown support too, but the inline links didn't seem to work as expected with v1.8.6, which we currently use on the build server. Besides, relative paths will be different on Github vs. iotdk.intel.com. This was mainly why I had to split the main readmes. I know it's an eyesore, so maybe I can try building the documentation with the latest Doxygen to see if links were fixed.

tony--, 2017-05-19:
Thanks for responding @Propanu.  I don't know if it's worth a big time investment.  It seemed like the original direction was to have iotdk.intel.com be the primary location for documentation to be read, in which case it makes sense to have it look good there.  However, in some places the docs actually point to Github, which makes me think reading on Github should be the priority.  

In any case, I just wanted to call it out and suggest what seemed like it could be a low-effort fix.  I'll close this one since you are already aware.

closed 2017-05-19 by tony--