eclipse-upm/upm #558: libupm-t6713.so does not link with libmraa.so

stefan-andritoiu, 2017-05-24:
Compiled t6713 example in ISS's docker:
i586-poky-linux-g++ --sysroot= -o t6713 src/main.o -lupm-t6713
/usr/lib/gcc/i586-poky-linux/4.9.1/../../../../i586-poky-linux/bin/ld: src/main.o: undefined reference to symbol 'mraa_i2c_stop'
/usr/lib/libmraa.so.1: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status

Did this and worked:
i586-poky-linux-g++ --sysroot= -o t6713 src/main.o -lupm-t6713 -lmraa

Propanu, 2017-05-31:
UPM examples should not have to link to MRAA directly, and rely only on the corresponding UPM library instead. There are a few such cases that need cleanup. What I'm surprised about though is the symbol that pops up, since the example only uses `mraa_get_sub_platform_id()`.

Propanu, 2017-06-01:
Is this the only C++ example you're having problems with?

stefan-andritoiu, 2017-06-02:
Only one so far, yes.