eclipse-upm/upm #559: Java sample code for the HTU21D returns a temperature of 0.0

alexwoolford, 2017-05-26:
The [HTU21DSample.java](https://github.com/intel-iot-devkit/upm/blob/master/examples/java/HTU21DSample.java) always returns a temperature of 0.0.

The following dependency was added to `pom.xml`:

    <dependency>
        <groupId>io.mraa.upm</groupId>
        <artifactId>upm_htu21d</artifactId>
        <version>1.2.0</version>
    </dependency>

I used the same values for the bus (1) and the I2C address (0x40; decimal value of 64) that [worked in Python](https://github.com/alexwoolford/htu21d_logger/tree/master/htu21d_logger_python). Here's a snippet of Java where I'm trying to read the temperature:

    upm_htu21d.HTU21D sensor = new upm_htu21d.HTU21D(1, 0x40);
    logger.info("sensor.getTemperature(): " + sensor.getTemperature());

The Java code returns a temperature of 0.0, where the Python code (linked above) shows the actual temperature. I experimented with different values for the bus and I2C address. Nothing worked.

Am I doing something wrong? Any suggestions for a (Java) workaround?

jontrulson, 2017-05-26:
Hi,
It looks like the Java example is missing a call to:

`sensor.sampleData();`

in the loop.  This is needed to actually read updated data from the device...

alexwoolford, 2017-05-26:
Thanks so much, Jon. That worked perfectly! I really appreciate your help.

stefan-andritoiu, 2017-05-26:
Yes, sorry about that. 
Fixed and restructured the Java example after the Python one https://github.com/intel-iot-devkit/upm/pull/562/commits/29e1489e40b8b8c5bcd0756725ae06eb9bf772d5

Propanu, 2017-06-01:
Merged to master. Thanks!

closed 2017-06-01 by Propanu