eclipse-upm/upm #560: ads1x15 fails to build with gcc-7

nullr0ute, 2017-05-26:
On Fedora 26 with gcc 7 I see the following failure due to more hardening flags

<pre>
cd /home/perobins/rpmbuild/BUILD/upm-1.3.0/src/ads1x15 && /usr/bin/c++  -Dads1x15_EXPORTS -I/home/perobins/rpmbuild/BUILD/upm-1.3.0/include -I/home/perobins/rpmbuild/BUILD/upm-1.3.0/src/ads1x15/.. -I/home/perobins/rpmbuild/BUILD/upm-1.3.0/src/interfaces -I/home/perobins/rpmbuild/BUILD/upm-1.3.0/src  -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -m64 -mtune=generic  -Wall -Wno-misleading-indentation -Wno-strict-aliasing -Werror -Wnon-virtual-dtor -Woverloaded-virtual -Wreorder -fPIC   -Wno-overloaded-virtual -std=gnu++11 -o CMakeFiles/ads1x15.dir/ads1x15.cxx.o -c /home/perobins/rpmbuild/BUILD/upm-1.3.0/src/ads1x15/ads1x15.cxx
/home/perobins/rpmbuild/BUILD/upm-1.3.0/src/ads1x15/ads1x15.cxx: In member function 'float upm::ADS1X15::getThresh(upm::ADS1X15::ADSTHRESH)':
/home/perobins/rpmbuild/BUILD/upm-1.3.0/src/ads1x15/ads1x15.cxx:119:25: error: enum constant in boolean context [-Werror=int-in-bool-context]
      if( THRESH_HIGH && THRESH_LOW) return getLastSample(reg);
                         ^~~~~~~~~~
/home/perobins/rpmbuild/BUILD/upm-1.3.0/src/ads1x15/ads1x15.cxx:119:25: error: enum constant in boolean context [-Werror=int-in-bool-context]
cc1plus: all warnings being treated as errors
make[2]: *** [src/ads1x15/CMakeFiles/ads1x15.dir/build.make:66: src/ads1x15/CMakeFiles/ads1x15.dir/ads1x15.cxx.o] Error 1
make[2]: Leaving directory '/home/perobins/rpmbuild/BUILD/upm-1.3.0'
make[1]: *** [CMakeFiles/Makefile2:2135: src/ads1x15/CMakeFiles/ads1x15.dir/all] Error 2
make[1]: Leaving directory '/home/perobins/rpmbuild/BUILD/upm-1.3.0'
make: *** [Makefile:133: all] Error 2
</pre>

Propanu, 2017-05-31:
Thanks for letting us know, I've fixed the logic with 8186d093ec1a2fca82ff15ea03dec86f3ca4c401!

closed 2017-05-31 by Propanu

nullr0ute, 2017-06-03:
This doesn't appear to fix the issue (patch against 1.3)

Propanu, 2017-06-03:
Sorry, pasted the wrong hash, it's c64d04d084f47b4cb3e1cdee6e96348fcc6a756e.

nullr0ute, 2017-06-03:
Looks good! Thanks :)