eclipse-upm/upm #568: upm/src/p9813/p9813.c:135

geedubess, 2017-06-08:
I'm not using this code, but I happened to do a google search for 'p9813 xor' while double-checking my own header algorithm, and found a bug in yours. Because I'm not using this code, I'm going to be lazy and not fork/issue a pull request. (:

        byte0 ^= (blue >> 2) & 0x30; // XOR bits 4-5
        byte0 ^= (green >> 4) & 0xC0; // XOR bits 2-3
        byte0 ^= (red >> 6) & 0x03; // XOR bits 0-1

0xC0 isn't bits 2-3. You want 0x0C for the green field mask to get bits 2-3!

malikabhi05, 2017-06-08:
Hi @geedubess , thanks for reporting this. It looks like there is a typo here, we will fix this.

malikabhi05, 2017-07-12:
This has been fixed and tested.

closed 2017-07-12 by malikabhi05