eclipse-upm/upm #569: Module version mismatch on all Node.js UPM modules

soldirered, 2017-06-08:
I am running a Siemens IoT2040 (which is basically a Galileo Gen 2) and Yocto Linux from the image provided by Siemens.
It came with Node.js 6.9.2, MRAA 1.5.1 and UPM 1.0.0
I upgraded MRAA and UPM to the most recent versions (1.7.0 and 1.3.0 respectively) via opkg and also with npm for the node bindings (npm also compiled everything again with node-gyp). Node.js is still at 6.9.2 (because it has been baked into the image and I don't want to create an image myself with bitbake)

However when I try to run my own code using a UPM library (or using one of the examples provided in the repository) I get a version mismatch. And code with MRAA functions runs just fine. 
Which kind of irritates me because everything has been rebuild by npm. 

The error message:
```javascript

root@iot2000:/usr/lib # node BMP280_SPI.js
module.js:597
  return process.dlopen(module, path._makeLong(filename));
                 ^

Error: Module version mismatch. Expected 48, got 46.
    at Error (native)
    at Object.Module._extensions..node (module.js:597:18)
    at Module.load (module.js:487:32)
    at tryModuleLoad (module.js:446:12)
    at Function.Module._load (module.js:438:3)
    at Module.require (module.js:497:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/usr/lib/BMP280_SPI.js:2:17)
    at Module._compile (module.js:570:32)
    at Object.Module._extensions..js (module.js:579:10)

```

And the code itself:
```javascript

require("mraa");
var sensorObj = require('jsupm_bmp280');

//Create new BMP280 sensor instance
var sensor = new sensorObj.BMP280(0, -1, 8);

//Get new informations from sensor
sensor.update();
	
//Print temperature and pressure to console	
console.log('Temperature: ' + sensor.getTemperature());
console.log('Pressure: ' + sensor.getPressure());

```

Any ideas?

Propanu, 2017-06-30:
Hi @soldirered, having both the opkg & npm versions on the same system is bound to break/cause issues. I would remove MRAA/UPM completely and then install the opkg version. If that doesn't work with the version of node on the board, you can try building our libraries from source with cmake. It's pretty straightforward. The npm packages work ok for MRAA, but UPM needs a bit more tweaking.

soldirered, 2017-07-02:
Hi @Propanu, thank you for the answer. I will try that out as soon as I can.