eclipse-upm/upm #573: ads1x15.cxx: fails to compile with gcc7

pohly, 2017-06-16:
```
float
ADS1X15::getThresh(ADSTHRESH reg){
     if( THRESH_HIGH && THRESH_LOW) return getLastSample(reg);
     else return 0.0;
}
```

fails to compile with gcc7 because THRESH_HIGH and THRESH_LOW are enums:

```
           /**
             * @enum ADSTHRESH
             * @brief uint8_t enum containing register addresses
             * used for setting HI and LOW threshold values as
             * well as setting conversion ready and set to default.
             *
             * @var ADSTHRESH::THRESH_LOW     = 0x02
             * @var ADSTHRESH::THRESH_HIGH    = 0x03
             * @var ADSTHRESH::CONVERSION_RDY = 0x04
             * @var ADSTHRESH::THRESH_DEFAULT = 0x05
             */
            typedef enum ADSTHRESH {
                THRESH_LOW      = ADS1X15_REG_POINTER_LOWTHRESH,
                THRESH_HIGH     = ADS1X15_REG_POINTER_HITHRESH,
                CONVERSION_RDY  = 0x04,
                THRESH_DEFAULT  = 0x05

            } ADSTHRESH;
```

Leading to:
```
ads1x15.cxx:119:25: error: enum constant in boolean context [-Werror=int-in-bool-context]
       if( THRESH_HIGH && THRESH_LOW) return getLastSample(reg);
                          ^~~~~~~~~~
ads1x15.cxx:119:25: error: enum constant in boolean context [-Werror=int-in-bool-context]
```

I don't know what the if check was meant to achieve. As it stands, older compilers probably just evaluated it to "true" at build time.

Propanu, 2017-06-16:
Hi, this was already fixed with c64d04d084f47b4cb3e1cdee6e96348fcc6a756e, and will be part of the next tag. See #560 also. Thanks!

Propanu, 2017-10-19:
Patch was merged and this should no longer be an issue with UPM 1.5.0

closed 2017-10-19 by Propanu