eclipse-upm/upm #586: UART sensor drivers with open(),  read() calls makes it not compatible with Android Things

vintummala, 2017-08-02:
There are few UART drivers which directly makes the open and read calls in the UPM driver instead of calling MRAA (which in turn calls the Peripheral manager for Android Things). With this, when these sensor drivers are used with Android Things it gives us permission denied. 
Android Things provides the security and makes it inevitable to go through peripheral manager calls. .

Below are the UPM sensor drivers which needs to be fixed to make it usable with Android Things are

1. zfm20: ---Fingerprint sensor module.
2. wt5001: --- This is a serial MP3 module.
3. scam:  -- Serial Camera module.
4. mhz16:  -- this is a gas sensor. This measures co2 and Temperature.
5. hmtrp: -- Serial RF PRO Module. This is for wireless communication. By default, the device sends and receives any data presented on its UART interface.
6. hm11:-- Bluetooth Low energy module.
7. grovescam: -- Grove serial camera.

The code snippet in the mhz sensor which is an issue with AT is as below .

MHZ16::MHZ16(int uart)
{
  m_ttyFd = -1;
  ..................
  // now open the tty
  if ( **(m_ttyFd = open(devPath, O_RDWR))** == -1)
    {
     ........
    }
}

int MHZ16::readData(char *buffer, int len)
{
 .........
  **int rv = read(m_ttyFd, buffer, len);**
 ..........
}

int MHZ16::writeData(char *buffer, int len)
{ ............
   .......
  **int rv = write(m_ttyFd, buffer, len);**
/write(                                   
  .........
}

pylbert, 2017-08-02:
I would hope these can be refactored to use the corresponding mraa::Uart::read/write methods instead of going straight to Linux tty devices.  Does anyone know the reasoning behind bypassing the MRAA api calls?

jontrulson, 2017-08-02:
Yes :)

At the time, MRAA did not have any real UART support.  The only call available was mraa_uart_init(), which simply setup the muxing so that the UART pins were tied to the appropriate linux UART device (/dev/tty*).

It wasn't until later that I added most of the current MRAA UART functionality, like read/write, data_available(), etc.  So these particular devices could be reworked to use the current existing MRAA UART functionality.

pylbert, 2017-08-07:
I converted 1 sensor class (ZFM20) as an example in this PR: https://github.com/intel-iot-devkit/upm/pull/587

If this looks OK, I can continue with the others.

vintummala, 2017-08-08:
This does not work for AT.  the dataAvailable() UPM api calls the uart_data_available_replace which is not implemented in peripheralman.c of MRAA. The reason for this is, AndroidThings peripheralmanager client only provides the polling method to veify the data availability. 
For this to work for AT, we need interrupt based calls in MRAA and also in UPM once implemented in MRAA.
@spitfire88 (Sanrio) could give more inputs on this.