eclipse-upm/upm #593: pyupm_bno055 module usage issue

karthikreddyk, 2017-08-17:
Hi, 

I am using above captioned module to capture IMU data from BNO055 , a 9 axis IMU from Bosch/Adafruit.
From time to time I get this message: bno055_read_reg: mraa_i2c_read_byte_data() failed, returning 0
and all of a sudden my accelerometer and angle outputs shoot up.

For example from say lets 1 m/s^2 to the X and Y component accel values displayed below:

accel_Data:
(-195.33999633789062, 137.60000610351562).

I am presuming the issue is because of lack of proper exception handling. Any help is greatly appreciated.
Sincerely,
Karthik Kambam

jontrulson, 2017-08-17:
Can you tell me what function was calling bno055_read_reg()?  Typically after you setup/calibrate the chip, you will then call update() in a loop, and then use the various getters, like getLinearAcceleration() to obtain the data of interest.

The update() routine is the only place this can happen during _normal_ operation, and the only thing it's used for in update() is obtaining the chip temperature.  So, during an update() call, the temperature would return 0.0 in this case.  It would not affect acceleration, gyro, or anything else (different low level routines are used for those).  So, I suspect it's just a symptom of another more serious problem.

We could fix this particular problem by modifying the code to check the result of every 'single register' read, and throw an exception, but the end result should still be the same in your case - since that routine is not used to obtain fusion and non-fusion data (accelerometer, etc)...

jontrulson, 2017-08-17:
I guess one other obvious question - are you using the chip temperature (via getTemperature()) for anything important in your code?

karthikreddyk, 2017-08-17:
Hi Jon,

Thank you for your prompt response. 
I am not calling getTemperature() at all.
In a loop I call update() first , then I call getLinearAcceleration(), and then getEulerAngles().
sometimes I get the message in sysout:
bno055_read_reg: mraa_i2c_read_byte_data() failed, returning 0.
In such cases, the acceleration data from getLinearAcceleration would be an impossible value like the one printed below:
accel_Data:
(-195.33999633789062, 137.60000610351562) - Z component not shown.
I am getting an impossible value from getEulerAngles() too.

The message: "bno055_read_reg: mraa_i2c_read_byte_data() failed, returning 0" makes me believe that the following module gets called when I call getLinearAcceleration() and getEulerAngles():

uint8_t bno055_read_reg(const bno055_context dev, uint8_t reg)
{
    assert(dev != NULL);
    int rv = mraa_i2c_read_byte_data(dev->i2c, reg);
    if (rv < 0)
    {
        printf("%s: mraa_i2c_read_byte_data() failed, returning 0\n",
               __FUNCTION__);
        return 0;
    }
    return (uint8_t)rv;
}
This is in https://raw.githubusercontent.com/intel-iot-devkit/upm/master/src/bno055/bno055.c

However you say that 
"and the only thing it's used for in update() is obtaining the chip temperature" 
If that were the case, can you please explain why I get "bno055_read_reg: mraa_i2c_read_byte_data() failed, returning 0" when calling getLinearAcceleration() and getEulerAngles() ?

Appreciate your help. Let me know if you need more information.

Sincerely,
Karthik Kambam

jontrulson, 2017-08-18:
> If that were the case, can you please explain why I get "bno055_read_reg: mraa_i2c_read_byte_data() failed, returning 0" when calling getLinearAcceleration() and getEulerAngles() ?

You can't, that's why I was asking :)  The failure is definitely in update() and the function paths it takes.

I can see how, if the i2c bus fails, you could get bad data.  What's probably happening is that bno055_read_reg() fails when reading the temperature.  This is not properly detected.  Then the other routines like _update_fusion_data() and _update_non_fusion_data() call bno055_read_regs() (which also fails, but does not emit an error message) then grabs your data, which is corrupt.

So, yes.  Error checking and propagation needs definite improvement in this driver.  I will mark myself as owner of this issue and see about submitting a PR to improve the situation.

In the short term though, you should try to find out why you are occasionally losing contact with the device (and/or bus).  The system logs might provide more information.

karthikreddyk, 2017-08-19:
Hi Jon,

I don't need temperature at all. Just FYI. May be you can write a separate update() for inertial functions alone. i.e Accel, Gyro and Magnetometers ? If it's not too much to ask :)

karthikreddyk, 2017-08-19:
the "system logs" might provide information. Where would these logs be ? I have connected the sensor to an Intel edison. Is the system logs, a folder in Linux Yocto installed on edison ? Please let me know. I don't understand how you are saying that there could be a loss of contact.

jontrulson, 2017-08-19:
The changes will be pervasive, so I can't just fix update() :(

As for the logs, try 'dmesg' for the kernel logs.  Also /var/log/messages or /var/log/syslog.  There should be some MRAA errors being reported.

jontrulson, 2017-08-19:
I've submitted PR #595 to correct the error detection and propagation issues you reported.

>Is the system logs, a folder in Linux Yocto installed on edison ?

For Edison, look at the output of _journalctl_, or run _journalctl -f_ in another shell while running your program.

>I don't understand how you are saying that there could be a loss of contact.

The reason these i2c errors are occurring is because the driver is occasionally having trouble talking to the device.  This would most likely be caused by unstable or broken connections to the device.  All this PR does is detect problems earlier and throw the appropriate exception on such an error, rather than return bad data.  You still need to fix whatever is causing your connection to the device to fail.  That can't be fixed in software...

karthikreddyk, 2017-08-21:
https://iotdk.intel.com/repos/3.5/intelgalactic/opkg/i586/ 
Hi Jon,
Initially I installed Pyupm library from above link. I checked the link to find that your update isn't made yet. When would you update your changes. Or am I missing something. Please let me know. Thank you.

jontrulson, 2017-08-21:
The maintainers need to merge or approve this PR before it can be merged into master.  After that, it will be whenever those packages are updated.

Propanu, 2017-08-29:
@karthikreddyk, until the next UPM tag, you can use the development repo to try it out: https://iotdk.intel.com/repos/3.5/intelgalactic-dev/opkg/i586/

Propanu, 2017-10-19:
UPM 1.5.0 includes this fix, closing. Thanks for the patch @jontrulson!

closed 2017-10-19 by Propanu