eclipse-upm/upm #597: Regression due to #812 in mraa

Mani-Sadhasivam, 2017-09-05:
mraa commit #812 introduces API change and fails to modify the corresponding drivers which in turn breaks the build. Following is the error observed while trying to build upm on Dragonboard410c.

`home/linaro/Documents/upm/src/kxcjk1013/kxcjk1013.cxx: In member function ��‘void upm::KXCJK1013::installISR(void (*)(char*), void*)��’:
/home/linaro/Documents/upm/src/kxcjk1013/kxcjk1013.cxx:70:45: error: invalid conversion from ��‘void (*)(char*)��’ to ��‘void (*)(char*, void*)��’ [-fpermissive]
     mraa_iio_trigger_buffer(m_iio, isr, NULL);
                                             ^
In file included from /home/linaro/Documents/upm/src/kxcjk1013/kxcjk1013.hpp:27:0,
                 from /home/linaro/Documents/upm/src/kxcjk1013/kxcjk1013.cxx:29:
/usr/include/mraa/iio.h:105:15: note:   initializing argument 2 of ��‘mraa_result_t mraa_iio_trigger_buffer(mraa_iio_context, void (*)(char*, void*), void*)��’
 mraa_result_t mraa_iio_trigger_buffer(mraa_iio_context dev, void (*fptr)(char*, void*), void* args);
               ^~~~~~~~~~~~~~~~~~~~~~~
[ 12%] Linking CXX shared library libupm-joystick12.so
src/kxcjk1013/CMakeFiles/kxcjk1013.dir/build.make:62: recipe for target 'src/kxcjk1013/CMakeFiles/kxcjk1013.dir/kxcjk1013.cxx.o' failed
make[2]: *** [src/kxcjk1013/CMakeFiles/kxcjk1013.dir/kxcjk1013.cxx.o] Error 1
CMakeFiles/Makefile2:14600: recipe for target 'src/kxcjk1013/CMakeFiles/kxcjk1013.dir/all' failed
make[1]: *** [src/kxcjk1013/CMakeFiles/kxcjk1013.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/home/linaro/Documents/upm/src/l3gd20/l3gd20.cxx: In member function ��‘void upm::L3GD20::installISR(void (*)(char*), void*)��’:
/home/linaro/Documents/upm/src/l3gd20/l3gd20.cxx:349:45: error: invalid conversion from ��‘void (*)(char*)��’ to ��‘void (*)(char*, void*)��’ [-fpermissive]
     mraa_iio_trigger_buffer(m_iio, isr, NULL);
                                             ^
In file included from /home/linaro/Documents/upm/src/l3gd20/l3gd20.hpp:31:0,
                 from /home/linaro/Documents/upm/src/l3gd20/l3gd20.cxx:31:
/usr/include/mraa/iio.h:105:15: note:   initializing argument 2 of ��‘mraa_result_t mraa_iio_trigger_buffer(mraa_iio_context, void (*)(char*, void*), void*)��’
 mraa_result_t mraa_iio_trigger_buffer(mraa_iio_context dev, void (*fptr)(char*, void*), void* args);
               ^~~~~~~~~~~~~~~~~~~~~~~
[ 12%] Linking CXX shared library libupm-l298.so
src/l3gd20/CMakeFiles/l3gd20.dir/build.make:62: recipe for target 'src/l3gd20/CMakeFiles/l3gd20.dir/l3gd20.cxx.o' failed
make[2]: *** [src/l3gd20/CMakeFiles/l3gd20.dir/l3gd20.cxx.o] Error 1
CMakeFiles/Makefile2:14829: recipe for target 'src/l3gd20/CMakeFiles/l3gd20.dir/all' failed
make[1]: *** [src/l3gd20/CMakeFiles/l3gd20.dir/all] Error 2
[ 12%] Built target joystick12
[ 12%] Built target l298
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
`

Mani-Sadhasivam, 2017-09-11:
@pylbert Are u aware of this?

dnoliver, 2017-09-11:
@Mani-Sadhasivam are you talking about this PR in mraa? https://github.com/intel-iot-devkit/mraa/pull/812

Propanu, 2017-09-11:
Hi @Mani-Sadhasivam, we are aware of the issue and working on updating the UPM IIO drivers to match the new API.

dnoliver, 2017-10-19:
@Propanu is this fixed?

Propanu, 2017-10-19:
Yes, closing.

closed 2017-10-19 by Propanu