eclipse-upm/upm #605: RE: Building/Compiling

silver2row, 2017-09-22:
Hello,

Here is my build error: 

**Install the project...
-- Install configuration: ""
-- Installing: /usr/local/include/mraa.h
CMake Error at src/cmake_install.cmake:36 (file):
  file INSTALL cannot copy file "/home/debian/mraa/api/mraa.h" to
  "/usr/local/include/mraa.h".
Call Stack (most recent call first):
  cmake_install.cmake:37 (include)

Makefile:72: recipe for target 'install' failed
make: *** [install] Error 1**

I am stuff for now in a plethora of ideas. What should my next move be? Aw!

Seth

Propanu, 2017-09-22:
`sudo make install`

silver2row, 2017-09-23:
Okay...So...

This is what I know so far, i.e. if this is any help at all.

I have a BBGW (BeagleBone Green Wireless) with a Debian Stretch Distro. I think I have my MRAA installed correctly because I can use the cmd sudo mraa-gpio list and get results. 

My machine belongs to me and I do not use anything but sudo instead of sudo -i. I do not want to make major mistakes. So...my folder outside of root is /home/debian as of now.

I have /home/debian/mraa instead of /home/debian/node_modules/mraa installed.

I used a couple of things to build the MRAA to what it is now. I used Python to see if the cmd line would work if I imported MRAA and then did a get Platform cmd to see if a board would appear. It showed BBB instead of the BBGW I am currently using. 

I followed info. from https://iotdk.intel.com/docs/master/upm/building.html and on this site under intel-iot-devkit building ideas. I have come up empty on both fronts. 

Does this help you help me?

Seth

P.S. Please reply when you have some extra time.

silver2row, 2017-09-23:
Hello Again,

I have not found out how to install UPM at all. Some sites state npm install jsupm_<module> and other sites state different things, e.g. cloning it from git.

What do you think I should do? I am at a loss. 

Seth

silver2row, 2017-09-23:
Hello Once More, 

I am so sorry for all this bugging about the UPM repository. I tried on the terminal:

sudo npm install jsupm_grovegprs.js -g

This did not work. 

I got a bunch of info. stating what I should do. I would post the .log file but it is lengthy.

Seth

P.S. It seems that the grovegprs.js file is not even in the upm/npm registry.

silver2row, 2017-09-23:
Hello...I got where I was wrong. I went to npmjs.org and found that I was incorrect by listening.

Seth

P.S. The info. I found from you was npm install jsupm_grovegprs and that is it. Thank you.

silver2row, 2017-09-23:
Hello,

Once I get the jsupm_grovegprs installed, do I need to compile/build it?

Seth

P.S. I have a .js file I use with node to run it on the terminal but I get errors. Um, do you have any recommendations?

silver2row, 2017-09-23:
Hello Again,

My CMakeLists.txt files are not located where I need them. How can I build without these CMakeLists.txt files?

Seth

silver2row, 2017-09-24:
I have a GPS module SIM28 and not a cellular module (GPRS). I think somewhere along the line, some of the  info. got changed incorrectly. See here: https://www.seeed.cc/BBGW-starter-tutorial%235-Where-are-you%3F-p-1648.html.

closed 2017-09-24 by silver2row

Propanu, 2017-10-10:
Hi @silver2row, we support the GPS too, see the **jsupm_nmea_gps** driver.