eclipse-upm/upm #619: Cannot use BMPX8X with UPM on Node.JS

urvoy-p, 2017-12-04:
Hi,

I am using Node.js 6.0.0 on a Raspberry Pi ZeroW running on Jessie.

The MRAA was tested and it works, other modules from UPM also work.

I am trying to read a BMP180 sensor, but I get this error. Seems like the node module cannot be instantiated.

There is the error I'm left with:

> home/pi/Projects/gardener/Sensors/build/index.js:20633
>         this.bmp = new UPM.BMPX8X();
>                    ^
> 
> Error: Illegal arguments for construction of _exports_BMPX8X
>     at Error (native)
>     at BMPX8X.setup (/home/pi/Projects/gardener/Sensors/build/index.js:20633:20)
>

pylbert, 2017-12-04:
Hello,

The 'Illegal arguments for construction ...' message is the default fall-through message from the SWIG generated wrapper which prints if the constructor for BMPX8X fails.  Looking at the C bmpx8x_init method, I would hope to see a printf to stdout which would include additional info on the error (however in your case above, I don't see this).

Looking at the bmpx8x_init method:

https://github.com/intel-iot-devkit/upm/blob/b1a49f0d3c3b543336b8fe6bf98b292da049acbc/src/bmpx8x/bmpx8x.c#L77-L95

the only case I see this return NULL w/o a printf seems to be if the malloc fails.  Can you use a debugger (gdb) to step through and find if/where the bmpx8x_init is failing?

pylbert, 2017-12-04:
Another thing to note for RPI, is the default I2C bus (0) from the BMPX8X C++ header will most likely need to be changed to the correct I2C bus for your RPI.  Maybe someone who uses RPI can chime in here (I will dig mine out and test but it may take a bit).  

In the meantime, mraa-i2c can print these out.  

Example:

$ mraa-i2c list
Bus   0: id=05 type=linux  default
Bus   1: id=04 type=linux

urvoy-p, 2017-12-04:
Thanks. Well I never bound gdb to node, should I setup a C++ project instead ?

Looking at the .c and .h , I don't see any reason for the malloc to fail.

And for the Raspberry Pi
> pi@raspberrypi:/usr/lib $ mraa-i2c list
> Bus   0: id=01 type=linux  default

pylbert, 2017-12-04:
The mraa tools and/or i2c-tools can be used to read a byte from the BMP180.  This can be helpful for debugging.

For your RPI, mraa I2C index 0 is I2C bus 1.  I believe either of these commands should read 0x55 from the CHIP ID register for the BMP180 (add sudo if needed).

mraa-i2c get 0 0x77 0xd0
i2cget -y 1 0x77 0xd0 b

WRT debugging - I often start a gdb session on the node interpreter, set a breakpoint in the upm source file/s, then run the gdb session.  You would need debug symbols for upm.

urvoy-p, 2017-12-05:
Yes the sensor is working:

> pi@raspberrypi:~/Sensors $ i2cget -y 1 0x77 0xd0 
> 0x55

I ran gdb on node:

> pi@raspberrypi:~/Sensors $ sudo gdb node
> (gdb) break ./node_modules/jsupm_bmpx8x/bmpx8x.cxx 
> (gdb)>[Make breakpoint pending on future shared library load? (y or [n]) y]
> (gdb) run build/index.js

Then the debug ends without entering the file.

You said that I need the symbols from upm (jsupm_bmpx8x). Do you somehow specify the .o files to gdb ?

I see they are located here for my project:

> /home/pi/Sensors/node_modules/jsupm_bmpx8x/build/Release/obj.target/jsupm_bmpx8x

>$ ls -l
> total 136
> -rw-r--r-- 1 pi pi  14596 Dec  4 07:42 bmpx8x.o
> -rw-r--r-- 1 pi pi 118304 Dec  4 07:42 jsupm_bmpx8xJAVASCRIPT_wrap.o
> drwxr-xr-x 2 pi pi   4096 Dec  4 07:41 utilities

urvoy-p, 2017-12-05:
Hi again,

I recompiled 'jsupm_bmpx8x' with 
> node-gyp rebuild --debug

Then I replaced 
`require('jsupm_bmpx8x');`
`require('../../node_modules/jsupm_bmpx8x/build/Debug/jsupm_bmpx8x.node');`

Ran the debugger as said before & still nothing

Can you please give me insight ?

I am planning to use UPM extensively, I would be really useful for me to be able debug.

Can you write a detailed procedure on how to debug with Node.js ? :)

pylbert, 2017-12-05:
@urvoy-p, try with python/java/c++ like @jontrulson recommends and hopefully that will shed some light on what's happening.

For your debugging reference... When debugging C/C++ UPM libraries you can use any of the language bindings (c++/nodejs/java/python).  The goal here is to debug the C/C++ UPM source via gdb, under the node interpreter (not debug the javascript source).

Assuming CMake (I'm building with CMake - I see you're using node-gyp), you can generate debug binaries which include symbols:

`$ cmake -DCMAKE_BUILD_TYPE=Debug -DBUILDSWIGNODE=on ..`
`$ make`
`$ cd build/src/bmpx8x`
`$ gdb node`
...
`(gdb) break bmpx8x.c:79`
...
`(gdb) run`
...
`> var x = require('jsupm_bmpx8x');`
...
`> x.BMPX8X()`
`[New Thread 0x7ffff2b3a700 (LWP 180771)]
Thread 1 "node" hit Breakpoint 1, bmpx8x_init (bus=0, addr=119) at tmp/upm_uart_mem_leak/src/bmpx8x/bmpx8x.c:79`
`(gdb)`

urvoy-p, 2017-12-06:
@pylbert I'm going to try that. Why is the error messages so poor on Node ?

pylbert, 2018-02-12:
@urvoy-p any luck with this issue?