eclipse-upm/upm #620: Failed to resolve: io.mraa.at  how to solve? or Can not more run android things apps in intel edison

neuberfran, 2017-12-07:
[image]

Propanu, 2017-12-08:
Hi @neuberfran, try changing your dependency to `compile 'io.mraa.at.upm:upm_jhd1313m1:1.+'`. Seems like our build that publishes Android Things packages for UPM timed out and only a few modules were updated to 1.5. We'll have to run it again.

neuberfran, 2017-12-08:
hi @Propanu  not solved. I had tried this
[image]

Propanu, 2017-12-08:
Let's try locking it to the previous version, 1.3.0 for the time being:
`compile 'io.mraa.at.upm:upm_jhd1313m1:1.3.0'`
I'll let you know once all packages have been updated to 1.5 and you should be able to switch back to your original code. There were no changes between versions to `jhd1313m1`.

neuberfran, 2017-12-08:
Ok. I stay waiting. But  the main driver I want to use is this:  compile 'io.mraa.at.upm:upm_smartdrive:1.+'
 from photo 02 below
[image]
[image]

Propanu, 2017-12-12:
Hi @neuberfran, I think most of the packages, including the ones you're trying to use have been updated by now.