eclipse-upm/upm #627: upm on raspberry pi 3

ayyappaswamy019, 2018-02-05:
Hello,
              currently iam working with upm on raspberry pi,while i was installing upm libray on raspberry pi the following error message coulld be displayed,libmraa was successufully installed.can you help me what is the problem in that....
src/led/CMakeFiles/led-c.dir/build.make:62: recipe for target 'src/led/CMakeFiles/led-c.dir/led.c.o' failed
make[2]: *** [src/led/CMakeFiles/led-c.dir/led.c.o] Error 1
CMakeFiles/Makefile2:15641: recipe for target 'src/led/CMakeFiles/led-c.dir/all' failed
make[1]: *** [src/led/CMakeFiles/led-c.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

thanks in advance....

pylbert, 2018-02-05:
@ayyappaswamy019 is there any additional info that you can provide on the error?  I don't see a reason for the failure in the txt above.

Propanu, 2018-02-05:
Hi @ayyappaswamy019, if you're building the UPM master branch from source you'll have to build against MRAA's master and not an installed 1.8.0.

ayyappaswamy019, 2018-02-07:
thanks for your reply......
pylbert, i mentioned the screen shot below check it once....
and i had done with the all the latest versions(libmraa and upm)
[image]

Propanu, 2018-02-07:
What compiler version are you using? (use `gcc -v`)
We have multiple such comparisons within UPM and I've never seen this warning before.
To suppress the warning, when you run cmake add the the following C flag:
`cmake -DCMAKE_C_FLAGS=-Wno-enum-compare ..`

ayyappaswamy019, 2018-02-08:
the following line shows my gcc version
gcc version 6.3.0 20170516 (Raspbian 6.3.0-18+rpi1)

and i used the above C flag but no use it shows same problem.....

ayyappaswamy019, 2018-02-08:
ok , only led module was giving those above error ,remaining all modules are working fine,now i have one more broblem....
if i enter this  below command 
pkg-config --cflags --libs upm-jhd1313m1
the following error was coming....
Package upm-jhd1313m1 was not found in the pkg-config search path.
Perhaps you should add the directory containing `upm-jhd1313m1.pc'
to the PKG_CONFIG_PATH environment variable
No package 'upm-jhd1313m1' found

closed 2018-02-22 by ayyappaswamy019

Propanu, 2018-02-22:
Sorry for the late reply, if you install under `/usr` or `/usr/local` pkg-config will find them automatically. With  a custom install directory you'll have to set the PKG_CONFIG_PATH variable accordingly. Also, the latest MRAA + UPM stable releases should fix the compilation error you were seeing.