eclipse-upm/upm #631: RuntimeError: UPM Runtime Error: BMM150: bmm150_init() failed

trixr4kdz, 2018-03-06:
Hi,

I currently have the Intel Aero RTF drone running Ubuntu 16.04 LTS and I wanted to be able to read the IMU data from the BMI160 chip that is built into the compute board, but I am stuck with initializing the device. The [Intel Aero guide](https://github.com/intel-aero/meta-intel-aero/wiki/94-(References)-Sensors-GPIOs-LEDs-CAN-bus#command-to-access-the-imu-sensor-on-the-aero-compute-board) says that the BMI160 chip uses bus 3 on chip 0.

Here is the simple code I'm trying to run currently based off the examples:
```python
#!/usr/bin/env python

from __future__ import print_function
import time, sys, signal, atexit
from upm import pyupm_bmi160 as sensorObj

sensor = sensorObj.BMI160(3, -1, 0, False)
```

And the error I am getting is the following:
```
bmi160_init: mraa_spi_init() failed.
Traceback (most recent call last):
  File "/home/trixie/imu_ws/src/aero_imu/scripts/test.py", line 7, in <module>
    sensor = sensorObj.BMI160(3, -1, 0, False)
  File "/usr/local/lib/python2.7/dist-packages/upm/pyupm_bmi160.py", line 161, in __init__
    this = _pyupm_bmi160.new_BMI160(bus, address, csPin, enableMag)
RuntimeError: UPM Runtime Error: BMI160: bmi160_init() failed
```

I also installed mraa and upm via ppa and there didn't seem to be any problems.

I am currently at a loss as to how to make this work so any help will be greatly appreciated.

Thanks!
Trixie

Propanu, 2018-03-06:
Hi, to use MRAA & UPM with the Intel Aero board you'll need to initialize MRAA using a JSON platform file. The `mraa.initJsonPlatform(std::string path) → mraa::Result` function would do it, but make sure to `import mraa` in your code too. I don't remember if the PPA version of the libraries have JSON platform support enabled, if this function call fails, you'll need to recompile MRAA and UPM from source. It's pretty straightforward and explained in our building documentation.

You can find the platform file for Intel Aero [here](https://github.com/intel-iot-devkit/mraa/blob/master/examples/platform/intel-aero.json).

Once initialized, SPI bus 3 is exposed as bus 0 in MRAA, thus your BMI160 constructor should look like this `sensor = sensorObj.BMI160(0, -1, -1, False)`. Setting the CS pin to -1 assumes it's hardware controlled.

Also, please note that the Intel Aero board supports Yocto officially. Ubuntu 16.04 might not expose the SPI bus in user space for MRAA to use, although there might be kernel patches available to work around this.

trixr4kdz, 2018-03-06:
Oh, I see. I missed that part from the mraa repository. I will check it out. Thank you so much!

Propanu, 2018-05-25:
Can you please post the MRAA log you get from `journalctl -f` or were you able to figure it out?

yehmostabsurd, 2019-06-14:
@trixr4kdz Did you figure this out? Is it possible to how you access BMI160?