eclipse-upm/upm #632: ds18b20 issue

cristina21r, 2018-03-14:
var sensor = new sensorObj.DS18B20(0);
             ^
Trying to run this code:

var sensorObj = require('jsupm_ds18b20');

/************** Main code **************/

console.log("Initializing...");

// Instantiate an DS18B20 instance using the default values (uart 0)
var sensor = new sensorObj.DS18B20(0);

console.log("Found", sensor.devicesFound(), "device(s)");
console.log("");

if (!sensor.devicesFound())
{
    process.exit(1);
}

// update and print available values every 2 seconds
setInterval(function()
{
    // update our values for all of the detected sensors
    sensor.update(-1);

    // we show both C and F for temperature for the sensors
    for (var i=0; i<sensor.devicesFound(); i++)
    {
        console.log("Device:", i, "Temperature:", sensor.getTemperature(i),
                    "C /", sensor.getTemperature(i, true), "F");
    }

    console.log();
}, 2000);

process.on('SIGINT', function()
{
    sensor = null;
    sensorObj.cleanUp();
    sensorObj = null;
    console.log("Exiting...");
    process.exit(0);
});

And i get this error:

Error: Illegal arguments for construction of _exports_DS18B20
    at Error (native)
    at Object.<anonymous> (/server/ds18.js:9:14)
    at Module._compile (module.js:409:26)
    at Object.Module._extensions..js (module.js:416:10)
    at Module.load (module.js:343:32)
    at Function.Module._load (module.js:300:12)
    at Function.Module.runMain (module.js:441:10)
    at startup (node.js:139:18)
    at node.js:968:3

Some help??

Propanu, 2018-03-14:
Hi @cristina21r, you need to use an external circuit with the ds18b20 sensor and it only works on UART pins. There's a handy diagram in our [known limitations](https://github.com/intel-iot-devkit/upm/blob/master/docs/knownlimitations.md#one-wire-sensors) section and you might find some additional details here: #421.

Propanu, 2018-05-25:
Were you able to get it to work?

Propanu, 2018-05-28:
What board are you trying this on, do you get any error messages from MRAA in the system log (`journalctl`)?