eclipse-upm/upm #636: UPM - SWIG 4.0 - .NET Core 2.1 C# - arm64v8

IvanFarkas, 2018-03-21:
Would it be possible to include **SWIG 4.0** to generate **.NET Core 2.1 C#** to use **UPM** on **arm64v8**, please?
Has anything done in this area?
If not, how can I help, please?

Propanu, 2018-03-21:
Currently being discussed on MRAA's issue tracker [here](https://github.com/intel-iot-devkit/mraa/issues/880).

IvanFarkas, 2018-03-21:
Yes. That's me too :-)

IvanFarkas, 2018-03-27:
I updated my  **[IvanFarkas/dotnetcore-arm-swig](https://github.com/IvanFarkas/dotnetcore-arm-swig)** GitHub repo with the help I received from @richlander and @Petermarcu at Microsoft.
It works great now using the **[new arm64 .NET Core 2.1 runtime](https://dotnetcli.blob.core.windows.net/dotnet/Runtime/master/dotnet-runtime-latest-linux-arm64.tar.gz)**.

I'd like to know how we can work together to provide **C# SWIG interface** for **MRAA** and **UPM**, please?
With that in place, a **C# .NET Core 2.1** app and an **Azure IoT Edge** module can access **IoT** sensors, actuators through the **MRAA** and **UPM** libraries. Exiting!