eclipse-upm/upm #640: Unable to compile bma220.py

nageshu, 2018-04-15:


pylbert, 2018-04-15:
@nageshu, make sure you [install ](https://github.com/intel-iot-devkit/upm/blob/master/docs/building.md#building-upm-------------------------building)after building the modules.

The CMake install command for UPM installs python2/3 modules under ${CMAKE_INSTALL_PREFIX}/${PYTHON2/3_PACKAGES_PATH}/upm.  The PYTHON2_PACKAGES_PATH can generally be site-packages or dist-packages depending on your distro.

You can grep this install path from your build directory.  For examples:

```
$ grep 'CMAKE_INSTALL_PREFIX' CMakeCache.txt
CMAKE_INSTALL_PREFIX:PATH=/usr/local

$ grep 'PYTHON.*PACKAGES' CMakeCache.txt
PYTHON2_PACKAGES_PATH:PATH=lib/python2.7/dist-packages
PYTHON3_PACKAGES_PATH:PATH=lib/python3.5/dist-packages
```

Would install to:

`/usr/local/lib/python2.7/dist-packages/upm`

pylbert, 2018-04-16:
@nageshu, most likely you need to install upm once you've built the project.  Try these steps (starting from the upm directory).

```
mkdir build
cd build
cmake ..
make
make install
```

Then open python and import the bma220 module:

```
$ python
Python 2.7.12 (default, Dec  4 2017, 14:50:18)
[GCC 5.4.0 20160609] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> from upm import pyupm_bma220
>>>

```

note - the above steps assume an install of [mraa](https://github.com/intel-iot-devkit/mraa).

nageshu, 2018-04-18:
Hey,
Thank you for replying my problem is solved now program is getting compiled.

But it is showing that invalid i2c bus and the path it is showing is as follows:
/usr/local/lib/python2.7/dist-package/upm/_pyupm_bma220.py 

In this file on line 330 it is showing error: Invalid i2c bus.
 
can you tell why it is showing this error?
and what method should I approach to overcome this error?
even if i am trying to make changes but i am unable to make changes. While saving program it is showing that permission denied.

Thanks In Advance!

nageshu, 2018-04-19:
Hey,
Thank you for replying,
I am using dragon board 410c having pre-installed linux in it. 
as you suggested I checked i2c bus list it is showing they are working
when I gone through the method you suggested it is showing following error

I have done following process:
linaro@linaro-alip:~/mraa/build/upm/examples/python$  mraa-i2c list
Bus   0: id=00 type=linux  default
Bus   1: id=00 type=linux
linaro@linaro-alip:~/mraa/build/upm/examples/python$ python
Python 2.7.13 (default, Nov 24 2017, 17:33:09)
[GCC 6.3.0 20170516] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> from upm import pyupm_bma220
>>> x= pyupm_bma220.BMA220(bus = 0, addr = 0xa)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/upm/pyupm_bma220.py", line 332, in __init__
    this = _pyupm_bma220.new_BMA220(bus, addr)
ValueError: UPM Invalid Argument: Unknown error in I2c::readReg()

please help me resolve this it will be really helpfull.

nageshu, 2018-04-24:
Hey,
can do i need to change register address according to my board?

pylbert, 2018-04-25:
@nageshu, I tested my up² with a bma220.  For reference, here's some I2C basics.

The mraa-i2c tool shows that the mraa index of '0' (mraa's default I2C bus) is the system I2C bus '5':

``` sh
$ mraa-i2c list
Bus   0: id=05 type=linux  default
Bus   1: id=04 type=linux
```

Doing a quick scan with mraa-i2c detect shows that there is a device on mraa I2C index '0' at 0x0a (the default I2C address of the bma220:

```
$ mraa-i2c detect 0
00: -- -- -- -- -- -- -- -- -- -- 0a -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
``` 

I also verified w/i2cdetect (from i2c-tools) to make sure the same addresses are returned (note, I2C system bus 5 == mraa I2C index 0) :

```
$ i2cdetect -y -r 5
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- 0a -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

Using the UPM python module I'm able to read the bma220's chip id:

```
$ python
Python 2.7.12 (default, Dec  4 2017, 14:50:18)
[GCC 5.4.0 20160609] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyupm_bma220
>>> x = pyupm_bma220.BMA220()
>>> hex(x.getChipID())
'0xdd'
```

If the above detect commands are not showing a device at address 0x0a, then the UPM module is not going to be able to read the device.  Here are some things to try:

1. Verify your wiring and power (my dfrobot bma220 has a (bright) blue LED)
2. Do you need sudo to access your IO?
3. Can you access any other devices w/I2C?
4. Pull-ups required?
5. Mraa will log its business to syslog (errors and info).  Take a look at your syslog for more information on the failure:

```
$ sudo grep mraa /var/log/syslog
...
Apr 25 02:58:02 UP-APL01 libmraa[18138]: libmraa version v1.9.0-24-g91fb838 initialised by user 'neck' with EUID 1001
Apr 25 02:58:02 UP-APL01 libmraa[18138]: Adding i2c bus found on i2c-5 on adapter i2c_designware.1
Apr 25 02:58:02 UP-APL01 libmraa[18138]: Adding i2c bus found on i2c-4 on adapter i2c_designware.0
Apr 25 02:58:02 UP-APL01 libmraa[18138]: up2: kernel pinctrl driver available
Apr 25 02:58:02 UP-APL01 libmraa[18138]: gpio: support for chardev interface is activated
Apr 25 02:58:02 UP-APL01 libmraa[18138]: libmraa initialised for platform 'UP2' of type 16
Apr 25 02:58:06 UP-APL01 libmraa[18138]: i2c_init: Selected bus 0

```

pylbert, 2018-05-02:
@nageshu, any progress?