eclipse-upm/upm #641: building.md doesn't specify that pyupm_utilities needs to be made as well if doing single module installs

vulq, 2018-04-17:
This took me hours to figure out, if not doing the full install but just selecting a single module (i.e. make _pyupm_bmi160-python3), you will get an error on import saying that the module doesn't exist (ImportError: No module named '_pyupm_bmi160'). To resolve this, you need to also install the utilities module (make _pyupm_utilities-python3).

Please include this in the build docs to help others who may have this issue.

pylbert, 2018-04-17:
@vulq, did you 'make install' after building your modules?  If so, can you provide more info and we can get this sorted out.

You can build/load a single module from your build-tree since binaries in the build tree have link dependencies specified with RPATH.  These are stripped upon install and rely on the run time linker/loader.

If you use 'make install,' the necessary dependencies will be installed.  Generally you should update the linker cache using 'ldconfig' before attempting to import a UPM python module - this could be added to the build docs.

If you are not using the UPM 'make install' mechanism to install the necessary content, you may need to do a small amount of hacking to get things working (e.g. ensure dependencies are in the correct places).  Note: using the build-tree binaries means you will have an RPATH pointing back to the build-tree (which can trip you up later on).

vulq, 2018-04-17:
I did do ldconfig and it didn't seem to have an effect. I didn't do a full make install since it was linking libraries that were not made; I was under the assumption, as per the docs, that I would only have to make install under build/src/bmi160 after doing make _pyupm_bmi160-python3. The reason the module could not be imported was because of a dependency on libupm-utilities.so. I feel like if this is a dependency for the bmi160 module, it should be included when building just a specific module - the docs seem to imply that all dependencies will be built.

pylbert, 2018-04-17:
@vulq, I see.  

The 'all' target under build/src/bmi160 comprehends building bmi160 (and _pyupm_bmi160-python3) and all necessary dependencies, but can only install components from this level.

I've added a note with #643.  Let me know if this is enough clarification.

vulq, 2018-04-17:
Great, thanks! So the only way to install those dependencies would be to do a make install in upm/build?

pylbert, 2018-04-17:
Correct.

I generally do all my building/installing from this level.  You can limit the number of directories CMake adds with the MODULE_LIST variable which helps to speed up generation and build time.

Example:

cmake -DMODULE_LIST=bmi160 ..
make -j4
make install

vulq, 2018-04-18:
Great, thank you so much!

Propanu, 2018-04-18:
You can actually go and install the dependencies manually from each `src` dir for the internal ones, and it would still be faster than running `make install` at top level in the `build` dir. The install target dependencies are missing from cmake and that's why a full installs work well. Having these would actually help package and distribute modules for python and node.js easier too.

pylbert, 2018-04-18:
@vulq, the doc updates are merged in 8f63914d76f8d6ff9971e3a97141012a2d78daf3.  OK to close?

closed 2018-04-18 by vulq