eclipse-upm/upm #646: ssd1306: darwPixel, drawLine etc

ric96, 2018-05-30:
right now the ssd1306 only seems to support drawing characters, whereas the eboled seems to be support darwPixel, drawLine etc being based on the same oled/lcd driver.
This issue is that eboled only supports spi and the 1306 supports i2c, it would be nice to either merge these two or at least have a common set of functions

Propanu, 2018-07-10:
Hi @ric96 I agree with you there's quite a bit of shared functionality between some of the displays supported in UPM, mainly the ones using the GFX header. I think there's a good opportunity to potentially review & merge this when we introduce the generic interfaces for the sensor classes in UPM. For this to work with the ssd1306 though, we would need a `drawPixel()` function that maps to the byte format needed by `draw()`.

aswinpajayan, 2024-11-14:
One possibility is to follow adafruits way of doing it .

I.e. collect all ssd 1306 realted commands into a class while seperating out the physical communication protocol(spi / i2c/ 3wire spi).

We can pass the pointer to the communication object while initialising the lcd *spi or *i2c 

upm::OLED(proto* .....
oled = new upm::OLED(*spi..

and modify OLED::command() to use 
proto->writeByte()

If this is acceptable. I can start workibg on this