eclipse-upm/upm #651: Accessing C++ enum from Python

DrNickDMartin, 2018-06-08:
I've been able to build and interact with my [Honeywell RSC](https://github.com/intel-iot-devkit/upm/tree/master/src/rsc) sensor using UPM (it's fantastic!). I've been using the Python bindings but I've run into a stumbling block.

I'm trying to set the mode and data rate for the sensor using the following:

```
from upm import pyupm_rsc as rsc
s = rsc.RSC(0,11,12)
mode = ? #Not sure what to do here
dr = ? #Not sure what to do here
s.setMode(mode)
s.setDataRate(dr)
```
However, I'm not sure how to access the `RSC_MODE` and `RSC_DATA_RATE` enums from the source code in when using Python. If I try passing an int to the function then I get a `TypeError`.

Any ideas?

Propanu, 2018-06-08:
Hi @DrNickDMartin, this is a shortcoming introduced when we switched to the generic SWIG interface files for the UPM modules. Enums and defines don't get picked up as constants in the SWIG generated wrapper if they are defined in a C header rather than a CPP one. @pylbert is working on a fix for it, thanks for pointing it out.

pylbert, 2018-06-08:
@DrNickDMartin, @Propanu, the pyupm_rsc module now exposes types from rsc_regs.h.  I added syntax to the python example which accesses these types.  While this code is non-functional (as far as the example is concerned), it would fail CI if these types are not exposed in the future.

These can be used like so:

https://github.com/intel-iot-devkit/upm/blob/3790c0bbc1482a5ce17f264feeaf567458378ccb/examples/python/rsc.py#L30-L31

closed 2018-06-08 by pylbert