eclipse-upm/upm #658: cmake Fail on https://github.com/intel-iot-devkit/upm

lenhuy, 2018-07-20:
OS: Ubuntu 16.04
Platform: upsquared-UP-APL01
Kernel: #5000-upboard11-Ubuntu SMP

cmake steps:
git clone https://github.com/intel-iot-devkit/upm.git
mkdir build
cd build
sudo apt install cmake
cmake ..

Error: CMakeFiles/cmTC_c054b.dir/CheckSymbolExists.c.o: In function `main':
          CheckSymbolExists.c:(.text+0x16): undefined reference to `pthread_create'

Log attached:
[CMakeError.log](https://github.com/intel-iot-devkit/upm/files/2214912/CMakeError.log)

Building C object CMakeFiles/cmTC_025b0.dir/CheckFunctionExists.c.o
/usr/bin/cc    -Wall -Wno-misleading-indentation -Wno-strict-aliasing -Wno-deprecated-declarations -Werror -Winit-self -Wimplicit -Wsign-compare -Wmissing-parameter-type -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTC_025b0.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-3.5/Modules/CheckFunctionExists.c

Propanu, 2018-07-20:
Hi @lenhuy, this is likely caused by missing dependencies. Most of them are present in the case of Ubuntu Desktop, but for Ubuntu Server you need to install some manually (e.g. `build-essential libc-dev`). Check the MRAA's building documentation for additional details, if you can get MRAA to build then UPM will do so too.

P.S. This can also be a cmake bug, but we'll discuss further if dependencies won't fix this for you.

pylbert, 2018-07-20:
I suspect the problem is not a missing thread implementation but one of the other dependencies like @Propanu mentioned.  

Can you provide the rest of the CMake output?  

Also, verify you have a 'libthread.so' (not libpthreads.so).

For reference...

The CMake find_package call for Threads creates multiple test snippet builds to determine a usable thread implementation.  It's normal to see errors in the CMakeError.log for the failing checks (it should *hopefully* find one and log that to CMakeOutput.log).  For example, my CMakeError.log looks nearly identical to yours (the pthread_create fails for two runs (1st, w/o a thread library specified and the 2nd with -lpthreads - note the 's' on pthreads - libpthread**s**.so prb doesn't exist on your ubuntu machine, but you should see a libpthread.so).

If I look in my CMakeOutput.**log**, I see the passing thread check:

```
Determining if the include file pthread.h exists passed with the following output:
Change Dir: /tmp/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_1dbd1/fast"
/usr/bin/make -f CMakeFiles/cmTC_1dbd1.dir/build.make CMakeFiles/cmTC_1dbd1.dir/build
make[1]: Entering directory '/tmp/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_1dbd1.dir/CheckIncludeFile.c.o
/usr/lib/ccache/cc     -o CMakeFiles/cmTC_1dbd1.dir/CheckIncludeFile.c.o   -c /tmp/build/CMakeFiles/CMakeTmp/CheckIncl
udeFile.c
Linking C executable cmTC_1dbd1
/usr/local/bin/cmake -E cmake_link_script CMakeFiles/cmTC_1dbd1.dir/link.txt --verbose=1
/usr/lib/ccache/cc       CMakeFiles/cmTC_1dbd1.dir/CheckIncludeFile.c.o  -o cmTC_1dbd1 -rdynamic·
make[1]: Leaving directory '/tmp/build/CMakeFiles/CMakeTmp'
```

lenhuy, 2018-07-20:
Thank you for quick response. 

1. I had build-essential libc-dev installed
2. mraa was built successfully
3. CMakeOutput.log is attached
[CMakeOutput.log](https://github.com/intel-iot-devkit/upm/files/2215339/CMakeOutput.log)

pylbert, 2018-07-20:
@lenhuy,

Can you provide the direct output from the cmake binary (everything after the call to 'cmake ..').

lenhuy, 2018-07-20:
upsquared@upsquared-UP-APL01:~/upm$ mkdir build
upsquared@upsquared-UP-APL01:~/upm$ cd build
upsquared@upsquared-UP-APL01:~/upm/build$ cmake ..
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Warnings as errors enabled (-Werror), disable with -DWERROR=off
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'mraa>=1.9.0'
--   Found mraa, version 1.9.0
-- Looking for mraa_iio_init in mraa
-- Looking for mraa_iio_init in mraa - found
-- Looking for mraa_firmata_init in mraa
-- Looking for mraa_firmata_init in mraa - not found
-- Looking for mraa_uart_ow_init in mraa
-- Looking for mraa_uart_ow_init in mraa - found
-- Checking for module 'libbacnet'
--   No package 'libbacnet' found
-- Checking for module 'libmodbus>=3.1.2'
--   
-- Checking for module 'libopenzwave'
--   No package 'libopenzwave' found
-- Could NOT find JPEG (missing:  JPEG_LIBRARY JPEG_INCLUDE_DIR) 
-- Found Nodejs: /usr/bin/node (found version "6.14.3") 
-- Found v8: /usr/include/node/v8.h (found version "5.1.281.111")
CMake Error at /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find SWIG (missing: SWIG_EXECUTABLE SWIG_DIR) (Required is at
  least version "3.0.5")
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindPackageHandleStandardArgs.cmake:388 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.5/Modules/FindSWIG.cmake:75 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:186 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/upsquared/upm/build/CMakeFiles/CMakeOutput.log".
See also "/home/upsquared/upm/build/CMakeFiles/CMakeError.log".
upsquared@upsquared-UP-APL01:~/upm/build$
[CMakeError.log](https://github.com/intel-iot-devkit/upm/files/2215455/CMakeError.log)
[CMakeOutput.log](https://github.com/intel-iot-devkit/upm/files/2215456/CMakeOutput.log)

Propanu, 2018-07-20:
@lenhuy, you are missing SWIG, you can solve this by running `sudo apt install swig`.

lenhuy, 2018-07-27:
test1.cpp:
#include "mraa/common.hpp"
#include "mraa/types.h"
#include "upm/light.hpp"

int main(int argc, char** argv)
{
    int upm::Light light(512);
    return 0;
}

g++ test1.cp -o test1 -lmraa -lupm-light
error: terminate called after throwing an instance of 'std::runtime_error'
    what(); Light:Light_init() failed.
Aborted (core dumped)

Where is the implementation of the constructor Light(unsigned int pin)?

lenhuy, 2018-07-31:
Need to be root user to access the GPIOs, so need to include sudo to run "sudo ./test1".

closed 2018-07-31 by lenhuy