eclipse-upm/upm #660: UPM build issue with gcc-8

b49020, 2018-08-06:
Hi,

I am facing UPM build issue (-DBUILDEXAMPLES=ON) with gcc-8 on Debian buster as follows:
```
[  223s] /usr/src/packages/BUILD/examples/c++/tm1637.cxx: In function 'int main(int, char**)':
[  223s] /usr/src/packages/BUILD/examples/c++/tm1637.cxx:69:29: error: '__builtin___snprintf_chk' output may be truncated before the last format character [-Werror=format-truncation=]
[  223s]          snprintf(myTime, 5, "%2d%02d", (hour + timezone + 24) % 24, min);
[  223s]                              ^~~~~~~~~
[  223s] In file included from /usr/include/stdio.h:862,
[  223s]                  from /usr/src/packages/BUILD/examples/c++/tm1637.cxx:26:
[  223s] /usr/include/aarch64-linux-gnu/bits/stdio2.h:64:35: note: '__builtin___snprintf_chk' output between 5 and 15 bytes into a destination of size 5
[  223s]    return __builtin___snprintf_chk (__s, __n, __USE_FORTIFY_LEVEL - 1,
[  223s]           ~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[  223s]         __bos (__s), __fmt, __va_arg_pack ());
[  223s]         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```
It seems that "format-truncation" flag in gcc-8 causes this build issue. So currently to avoid this issue I have applied following patch in UPM:

```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 89b0ed31..982e0e03 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -91,7 +91,8 @@ endfunction ()
 # Compiler flags common to both C and CXX
 # Enable -Wall
 # GCC-6 added -Wmisleading-indentation to -Wall, skip these for now
-set (C_CXX_WARNING_FLAGS -Wall -Wno-misleading-indentation -Wno-strict-aliasing)
+# GCC-8 added -Wformat-truncation to -Wall, skip these for now
+set (C_CXX_WARNING_FLAGS -Wall -Wno-misleading-indentation -Wno-strict-aliasing -Wno-format-truncation)
 
 # Warnings as errors?
 if (WERROR)
```
Please advise me if above patch is acceptable or do we have a proper fix for this format-truncation issue.

Regards,
Sumit

b49020, 2018-08-07:
@pylbert Actually my build stopped with this error only. I would try to replace C snprintf methods in the C++ example with std::cout and build further.

pylbert, 2018-08-07:
@b49020, I imagine you'll encounter more.

You can always tell make to continue and see how many more there are:

```make -k```

b49020, 2018-08-07:
@pylbert Following is one more format-truncation error but here its a "c" example, so replacement won't work. Apart from these two I don't see any.
```
[  807s] /usr/src/packages/BUILD/examples/c/jhd1313m1.c: In function 'main':
[  807s] /usr/src/packages/BUILD/examples/c/jhd1313m1.c:65:49: warning: '%d' directive output may be truncated writing between 1 and 10 bytes into a region of size 8 [-Wformat-truncation=]
[  807s]          snprintf(str, sizeof(str), "Hello World %d", ndx);
[  807s]                                                  ^~
[  807s] /usr/src/packages/BUILD/examples/c/jhd1313m1.c:65:36: note: directive argument in the range [0, 2147483647]
[  807s]          snprintf(str, sizeof(str), "Hello World %d", ndx);
[  807s]                                     ^~~~~~~~~~~~~~~~
[  807s] In file included from /usr/include/stdio.h:862,
[  807s]                  from /usr/src/packages/BUILD/src/jhd1313m1/jhd1313m1.h:32,
[  807s]                  from /usr/src/packages/BUILD/examples/c/jhd1313m1.c:25:
[  807s] /usr/include/aarch64-linux-gnu/bits/stdio2.h:64:10: note: '__builtin___snprintf_chk' output between 14 and 23 bytes into a destination of size 20
[  807s]    return __builtin___snprintf_chk (__s, __n, __USE_FORTIFY_LEVEL - 1,
[  807s]           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[  807s]         __bos (__s), __fmt, __va_arg_pack ());
[  807s]         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

pylbert, 2018-08-07:
@b49020, thanks for looking into this.  The warning seems valid since the %d can format up-to 10 characters which would overrun str if the while loop is allowed to run (a long time).  

I'm not at a place where I can test this, but I imagine it would be possible to add length specifiers to the format string for snprintf or increase the size of the str array to a valid value (23?).  Can you test this?

b49020, 2018-08-09:
@pylbert Raised a pull request #663 to fix above issues.

Propanu, 2018-09-07:
Merged and should be fixed. Thank you!

closed 2018-09-07 by Propanu