eclipse-upm/upm #671: CMake Error at src/CMakeLists.txt:929 (add_subdirectory)

mrsandeshbhat, 2018-10-03:
I have been trying to cmake upm for some days now. The most recent error log and cmake output files are attached below:
[CMakeOutput.log](https://github.com/intel-iot-devkit/upm/files/2444271/CMakeOutput.log)
[CMakeError.log](https://github.com/intel-iot-devkit/upm/files/2444272/CMakeError.log)

Also, the terminal output is as follows:

cmake .. -DBUILDSWIGNODE=OFF
-- The C compiler identification is GNU 4.9.2
-- The CXX compiler identification is GNU 4.9.2
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Warnings as errors enabled (-Werror), disable with -DWERROR=off
-- Performing Test HAS__Wall
-- Performing Test HAS__Wall - Success
-- Performing Test HAS__Wno_misleading_indentation
-- Performing Test HAS__Wno_misleading_indentation - Success
-- Performing Test HAS__Wno_strict_aliasing
-- Performing Test HAS__Wno_strict_aliasing - Success
-- Performing Test HAS__Wno_deprecated_declarations
-- Performing Test HAS__Wno_deprecated_declarations - Success
-- Performing Test HAS__Werror
-- Performing Test HAS__Werror - Success
-- Performing Test HAS__Winit_self
-- Performing Test HAS__Winit_self - Success
-- Performing Test HAS__Wimplicit
-- Performing Test HAS__Wimplicit - Success
-- Performing Test HAS__Wsign_compare
-- Performing Test HAS__Wsign_compare - Success
-- Performing Test HAS__Wmissing_parameter_type
-- Performing Test HAS__Wmissing_parameter_type - Success
-- Performing Test HAS__Wnon_virtual_dtor
-- Performing Test HAS__Wnon_virtual_dtor - Success
-- Performing Test HAS__Woverloaded_virtual
-- Performing Test HAS__Woverloaded_virtual - Success
-- Performing Test HAS__Wreorder
-- Performing Test HAS__Wreorder - Success
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.28")
-- checking for module 'mraa>=2.0.0'
--   found mraa, version 2.0.0
-- Looking for mraa_iio_init in mraa
-- Looking for mraa_iio_init in mraa - found
-- Looking for mraa_firmata_init in mraa
-- Looking for mraa_firmata_init in mraa - not found
-- Looking for mraa_uart_ow_init in mraa
-- Looking for mraa_uart_ow_init in mraa - found
-- checking for module 'libbacnet'
--   package 'libbacnet' not found
-- checking for module 'libmodbus>=3.1.2'
--   package 'libmodbus>=3.1.2' not found
-- checking for module 'libopenzwave'
--   package 'libopenzwave' not found
-- Could NOT find JPEG (missing:  JPEG_LIBRARY JPEG_INCLUDE_DIR)
-- Found SWIG: /usr/local/bin/swig (found suitable version "3.0.5", minimum required is "3.0.5")
-- Found PythonInterp: /usr/bin/python2.7 (found suitable version "2.7.9", minimum required is "2.7")
-- Found PythonLibs: /usr/lib/i386-linux-gnu/libpython2.7.so (found suitable exact version "2.7.9")
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.4.2", minimum required is "3")
-- Found PythonLibs: /usr/lib/i386-linux-gnu/libpython3.4m.so (found suitable exact version "3.4.2")
-- Building python2 modules with python-2.7.9
-- Building python3 modules with python-3.4.2
-- UPM Version v1.7.0-4-g85608a9
-- Target arch is i586
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Performing Test COMPILER_SUPPORTS_C11
-- Performing Test COMPILER_SUPPORTS_C11 - Success
-- Performing Test HAS__Wno_error
-- Performing Test HAS__Wno_error - Success
-- Performing Test HAS__Wno_delete_non_virtual_dtor
-- Performing Test HAS__Wno_delete_non_virtual_dtor - Success
-- Performing Test HAS__Wno_unused_function
-- Performing Test HAS__Wno_unused_function - Success
-- Performing Test HAS__Wno_maybe_uninitialized
-- Performing Test HAS__Wno_maybe_uninitialized - Success
**CMake Error at src/CMakeLists.txt:929 (add_subdirectory):
  The binary directory

    /root/upm/build/src/utilities

  is already used to build a source directory.  It cannot be used to build
  source directory

    /root/upm/src/utilities

  Specify a unique binary directory name.**

-- Performing Test HAS__Wno_overloaded_virtual
-- Performing Test HAS__Wno_overloaded_virtual - Success
-- UPM target summary:
--      Found C       libraries: 94
--      Found CXX     libraries: 217
--      Found JAVA    libraries: 0
--      Found NODEJS  libraries: 0
--      Found PYTHON2 libraries: 217
--      Found PYTHON3 libraries: 217
-- Generating pkg-config files for 94 C libraries...
-- Generating pkg-config files for 94 C libraries... Complete
-- Generating pkg-config files for 217 CXX libraries...
-- Generating pkg-config files for 217 CXX libraries... Complete
-- Configuring incomplete, errors occurred!
See also "/root/upm/build/CMakeFiles/CMakeOutput.log".
See also "/root/upm/build/CMakeFiles/CMakeError.log".

Any suggestions will be appreciated.

Propanu, 2018-10-11:
Hi @mrsandeshbhat this might be caused by a dirty build directory. Do you have any files in it? I recommend deleting it and creating it again before running cmake.

mrsandeshbhat, 2018-10-25:
Hello @Propanu I tried deleting the upm folder using (rm -R upm) and running cmake again. I am getting the same error. Do you have any idea as to why this might be happening?

mrsandeshbhat, 2018-10-28:
Fixed it!!

Before the solution, here is what I tried and failed at:
1. delete the build directory and cmake again - failed.
2. delete the upm directory, git the upm directory and cmake again - failed.
3. re-flash the jubilinux OS and try all that again - failed.

Now here is what worked:

I flashed a ubilinux OS on my board and then reflashed it with the jubilinux OS and tried it again. 

The cmake seems to fail if the OS had some of the dirty files from a previous build attempt and even a fresh OS install install didn't help it. But now it works.