eclipse-upm/upm #673: FTBFS error: 'WeakCallbackData' in namespace 'v8' does not name a template type

thiagomacieira, 2018-10-09:
Build steps:
```
mkdir build
cmake -G 'Unix Makefiles' -DCMAKE_INSTALL_PREFIX=/usr -DBUILD_SHARED_LIBS:BOOL=ON -DLIB_INSTALL_DIR:PATH=/usr/lib64 -DLIB_SUFFIX=64 -DCMAKE_AR=/usr/bin/gcc-ar -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_RANLIB=/usr/bin/gcc-ranlib .. -DWERROR=OFF
make
```

Relevant CMake output:
```
-- Found JPEG: /usr/lib64/libjpeg.so (found version "62") 
-- Found Nodejs: /usr/bin/node (found version "8.12.0") 
-- Found v8: /usr/include/node/v8.h (found version "6.2.414.66")
-- Found SWIG: /usr/bin/swig (found suitable version "3.0.12", minimum required is "3.0.5") 
-- Found PythonInterp: /usr/bin/python2.7 (found suitable version "2.7.15", minimum required is "2.7") 
-- Found PythonLibs: /usr/lib64/libpython2.7.so (found suitable exact version "2.7.15") 
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.7", minimum required is "3") 
-- Found PythonLibs: /usr/lib64/libpython3.7m.so (found suitable exact version "3.7.0") 
-- Building python2 modules with python-2.7.15
-- Building python3 modules with python-3.7.0
-- UPM Version v1.7.0
CMake Warning at CMakeLists.txt:233 (message):
  Failed to retrieve UPM version with 'git describe' (using v1.7.0).  Check
  that git is installed and this is a valid git repo.
-- Target arch is x86_64
-- UPM target summary:
--      Found C       libraries: 94
--      Found CXX     libraries: 218
--      Found JAVA    libraries: 0
--      Found NODEJS  libraries: 218
--      Found PYTHON2 libraries: 218
--      Found PYTHON3 libraries: 218
```

Results in
```
adafruitssJAVASCRIPT_wrap.cxx:966:27: error: 'WeakCallbackData' in namespace 'v8' does not name a template type
   void (*dtor) (const v8::WeakCallbackData<v8::Object, SWIGV8_Proxy> &data);
                           ^~~~~~~~~~~~~~~~
adafruitssJAVASCRIPT_wrap.cxx:966:23: note: suggested alternative: 'WeakCallbackType'
   void (*dtor) (const v8::WeakCallbackData<v8::Object, SWIGV8_Proxy> &data);
                       ^~
                       WeakCallbackType
```
Plus a couple thousand more errors.

Propanu, 2018-10-11:
MRAA/UPM won't build with nodejs 7.0.0+ out of the box, because of a missing SWIG patch. The workaround is described on the MRAA documentation pages: https://github.com/intel-iot-devkit/mraa/blob/master/docs/building.md#javascript-bindings-for-nodejs-700
Alternatively, you could simply disable the node bindings with `-DBUILDSWIGNODE=OFF`.

kapilbhudhia, 2021-07-01:
the patch suggested in the above link is not available anymore(?)