eclipse-upm/upm #675: Query: Is there any plan to support BME680 Bosch Sensor

BinduRao2018, 2019-02-07:
We plan to use BME680 instead of BME280 and would like to know if there is any plan to add this sensor support?

Propanu, 2019-02-19:
Hi @BinduRao2018, we are not currently working on adding new Bosch/Grove sensors to the UPM library and are instead focused on consolidating the APIs based on sensor categories for the next major update. If you are actively using this sensor and would like to try and find help for porting the library you linked to MRAA supported boards please let us know.

BinduRao2018, 2019-02-28:
Hi @Propanu,
I managed to integrate bme680, using https://github.com/BoschSensortec/BME680_driver as a reference. The functionality provided from my work uses mraa APIs and not UPM.  This was verified on UP2 Grove IoT Development Kit and Raspberry Pi.  W.r.t outputs from the sensors for Temp, Humidity and Pressure, I would say it was fairly reasonable and not very accurate. This is due to the calibration values (I have used the same reference values as available in the above mentioned repo) as the details are not available. 
PS: I haven't integrated the changes with BSEC library that is suppose to provide more accurate measurement values, however the library is not available on all platforms for download.