eclipse-upm/upm #676: Error while building upm on raspberry pi3

prags123, 2019-02-11:
/home/upm/build $ sudo cmake -DCMAKE_INSTALL_PREFIX:PATH=/usr -DCMAKE_BUILD_TYPE=DEBUG -DCMAKE_C_COMPILER=/usr/bin/clang -CMAKE_CXX_COMPILER=/usr/bin/clang++ ..

On running cmake command got the below mentioned error - 

loading initial cache file MAKE_CXX_COMPILER=/usr/bin/clang++
CMake Error: Error processing file: MAKE_CXX_COMPILER=/usr/bin/clang++
-- The C compiler identification is Clang 3.8.1
-- The CXX compiler identification is GNU 6.3.0
-- Check for working C compiler: /usr/bin/clang
-- Check for working C compiler: /usr/bin/clang -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Warnings as errors enabled (-Werror), disable with -DWERROR=off
CMake Warning at CMakeLists.txt:84 (message):
  C compiler does not support flag "-Wno-misleading-indentation"
Call Stack (most recent call first):
  CMakeLists.txt:109 (upm_add_compile_flags)

CMake Warning at CMakeLists.txt:84 (message):
  C compiler does not support flag "-Wmissing-parameter-type"
Call Stack (most recent call first):
  CMakeLists.txt:109 (upm_add_compile_flags)

CMake Warning at CMakeLists.txt:84 (message):
  CXX compiler does not support flag "-Wno-misleading-indentation"
Call Stack (most recent call first):
  CMakeLists.txt:117 (upm_add_compile_flags)

-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29") 
-- Checking for module 'mraa>=2.0.0'
--   
CMake Error at /usr/share/cmake-3.7/Modules/FindPkgConfig.cmake:415 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.7/Modules/FindPkgConfig.cmake:588 (_pkg_check_modules_internal)
  CMakeLists.txt:134 (pkg_check_modules)

-- Configuring incomplete, errors occurred!
See also "/home/upm/build/CMakeFiles/CMakeOutput.log".
See also "/home/upm/build/CMakeFiles/CMakeError.log".

please help.

Propanu, 2019-02-19:
Hi @prags123, sorry for the delay. The UPM project requires MRAA in order to compile and function correctly. And while MRAA is distributed natively for many Linux distributions, you might have to compile it too for Raspbian (if you're running Ubuntu on your RPi check out the PPA we provide as it might work directly). Build instructions for MRAA can be found on the project page [mraa.io](https://mraa.io), and there's also a good guide specifically for RPi boards [here](https://www.ics.com/blog/getting-started-mraa-raspberry-pi).

prags123, 2019-02-19:
hello Propanu, I know that The UPM project requires MRAA in order to compile and function correctly. And while MRAA is distributed natively for many Linux distributions, you might have to compile it too for Raspbian (if you're running Ubuntu on your RPi check out the PPA we provide as it might work directly).
I m running raspbian 9 stretch on RPI and trying to install upm after building MRAA library,but it shows error - its is unable to find mraa library.can u plz guide what is the issue ? How i uninstall the previous version of MRAA from raspbian stretch 9.

g-vidal, 2019-04-12:
Hi,
I am compiling mraa and upm on a raspberry Pi with a true debian stretch or buster and I cannot reproduce your problem. Would-it be linked with some raspbian tricks?