eclipse-upm/upm #677: rfid pn532 invalid gpio with up2squared board

Orch3strator, 2019-02-15:
Hello,

I've got a rfid board PN532 connected to the I2C bus via the groove board and directly to the GPIO ping 3 and 5. Here's what I was able to get to so far:
Connected directly to I2C:
**nfc-scan-device -v**
nfc-scan-device uses libnfc 1.7.1
1 NFC device(s) found:

- pn532_i2c:/dev/i2c-5:
    pn532_i2c:/dev/i2c-5
chip: PN532 v1.6
initator mode modulations: ISO/IEC 14443A (106 kbps), FeliCa (424 kbps, 212 kbps), ISO/IEC 14443-4B (106 kbps), Innovision Jewel (106 kbps), D.E.P. (424 kbps, 212 kbps, 106 kbps)
target mode modulations: ISO/IEC 14443A (106 kbps), FeliCa (424 kbps, 212 kbps), D.E.P. (424 kbps, 212 kbps, 106 kbps)

**nfc-list**
nfc-list uses libnfc 1.7.1
NFC device: pn532_i2c:/dev/i2c-5 opened
1 ISO14443A passive target(s) found:
ISO/IEC 14443A (106 kbps) target:
    ATQA (SENS_RES): 00  04
       UID (NFCID1): cb  c2  94  0b
      SAK (SEL_RES): 08

If' I connect the pn532 via the groove shield I get nothing. If I scan the I2C bus first, I get nothing (have to reboot the machine to get the nfc commands working again.)

Using the python example with mraa/upm I get this error message:
Example from: https://github.com/intel-iot-devkit/upm/blob/master/examples/python/pn532.py

**python rfid.py**
Traceback (most recent call last):
  File "rfid.py", line 68, in <module>
    main()
  File "rfid.py", line 11, in main
    myNFC = upmPn532.PN532(3, 2)
  File "/usr/lib/python2.7/dist-packages/upm/pyupm_pn532.py", line 246, in __init__
    this = _pyupm_pn532.new_PN532(*args)
ValueError: UPM Invalid Argument: Invalid GPIO pin specified

**python3 rfid.py**
Traceback (most recent call last):
  File "rfid.py", line 68, in <module>
    main()
  File "rfid.py", line 11, in main
    myNFC = upmPn532.PN532(3, 2)
  File "/usr/lib/python3.5/dist-packages/upm/pyupm_pn532.py", line 246, in __init__
    this = _pyupm_pn532.new_PN532(*args)
ValueError: UPM Invalid Argument: Invalid GPIO pin specified

I've changed to pin 3 and 5 according to https://wiki.up-community.org/Pinout_UP2 to no avail.

Any suggestions on how to get this to work with mraa/upm and python?
mraa version installed: 2.0.0.-2
upm version installed: 1.7.1-1

Regards, O.

Propanu, 2019-02-15:
Hi @Orch3strator, unfortunately the UPM Python documentation doesn't include descriptions for the constructor unless you expand the source and that's easy to overlook. However, if you scroll to the __init__ function [here](http://iotdk.intel.com/docs/master/upm/python/_modules/pyupm_pn532.html#PN532) you will notice that 4 arguments are used to successfully initialize the PN532 modules: IRQ pin, RST pin, I2C bus, I2C address. Implicit values are used for the I2C bus, but in your case I think you'll have to use the full constructor. The 2 pin values you changed in the example are not the I2C bus, rather separate GPIOs required to use the module.

Orch3strator, 2019-02-20:
Thank you @Propanu. I'm using the GrovePi board and added to my python script:
```
def initGrove():
	    #The UP^2 treats the shield as a sub platform over I2C 
	    mraa.addSubplatform(mraa.GROVEPI, "0")

initGrove()  
```

However, I can't figure out what to use for: IRQ pin, RST pin, I2C bus, I2C address.
My libnfc.con points to:
device.name = "PN532 over I2C"
device.connstring = "pn532_i2c:/dev/i2c-5"

Appreciate your help. Regards, O.