eclipse-upm/upm #681: C++ API and Abstract classes

abogani, 2019-04-10:
Why doesn't UPM use the abstract classes to improve code portability between devices (i.e. for Stepper Motors)? Is it a choice made by design? If yes why?

Propanu, 2019-04-14:
Hi @abogani, and thanks for raising this question. We do include a good number of interfaces to help with portability for sensors of the same type. About 10 of these have been around for quite some time in order to collect developer feedback and determine what's the best way to handle them via SWIG. We have since increased the number to over 30 and now, many UPM sensors implement them. The best part about this is that the new interfaces are useful even with the other languages (Java, JS & Python).

Actuators are a bit more complicated and this is were we continue to ask for developer feedback and help. Let's use the stepper motor as the example. There are 2 popular generic implementations (step and direction or "direct" coil control via gpio). Add to this hundreds of very custom industrial controllers or weird combinations of the generic methods. I could easily say that step and dir functions will make for a good interface for stepper motors, but then you will come in and say speed control is also very important and should be part of the abstraction rather than user code. And this is where it gets challenging, because setting the speed needs utility classes and the implementation will now vary substantially based on the target device, going against what the abstraction was meant to do.

So to summarize, we did abstract a lot of sensors types, but actuators are a different story and we invite the community to take the initiative for them. Hope this helps clarify things but please let us know if you have additional questions or suggestions on this!

abogani, 2019-04-15:
@Propanu Thanks for detailed explanation!

What do you suggest me to do then? Should I suggest code of an asbtract class for stepper motors or should I implement an abstraction layer for that type of actuators at user level code (my application)?

Thanks again!

Propanu, 2019-04-19:
Hi @abogani, sorry for the delay. If you want to open a PR with a new iStepper interface, I will be more than happy to help shape it and get it integrated with the rest of the UPM. With 2 pure virtual functions for step and direction it should be possible to cover most if not all of the steppers in the project.