eclipse-upm/upm #692: FTBFS with Noje.js 12: error: no matching function for call to ‘v8::Value::ToObject()’

thiagomacieira, 2019-11-11:
```
make[2]: Entering directory '/builddir/build/BUILD/upm-2.0.0/clr-build'
/builddir/build/BUILD/upm-2.0.0/clr-build/interfaces/CMakeFiles/jsupm_interfaces.dir/interfacesJAVASCRIPT_wrap.cxx: In function ‘int SWIG_V8_GetInstancePtr(v8::Handle<v8::Value>, void**)’:
/builddir/build/BUILD/upm-2.0.0/clr-build/interfaces/CMakeFiles/jsupm_interfaces.dir/interfacesJAVASCRIPT_wrap.cxx:1030:52: error: no matching function for call to ‘v8::Value::ToObject()’
 1030 |   v8::Handle<v8::Object> objRef = valRef->ToObject();
      |                                                    ^
In file included from /usr/include/node/node.h:63,
                 from /builddir/build/BUILD/upm-2.0.0/clr-build/interfaces/CMakeFiles/jsupm_interfaces.dir/interfacesJAVASCRIPT_wrap.cxx:171:
/usr/include/node/v8.h:2576:44: note: candidate: ‘v8::MaybeLocal<v8::Object> v8::Value::ToObject(v8::Local<v8::Context>) const’
 2576 |   V8_WARN_UNUSED_RESULT MaybeLocal<Object> ToObject(
      |                                            ^~~~~~~~
/usr/include/node/v8.h:2576:44: note:   candidate expects 1 argument, 0 provided
In file included from /usr/include/node/v8-internal.h:14,
                 from /usr/include/node/v8.h:25,
                 from /usr/include/node/node.h:63,
                 from /builddir/build/BUILD/upm-2.0.0/clr-build/interfaces/CMakeFiles/jsupm_interfaces.dir/interfacesJAVASCRIPT_wrap.cxx:171:
/usr/include/node/v8.h:2590:31: note: candidate: ‘v8::Local<v8::Object> v8::Value::ToObject(v8::Isolate*) const’
 2590 |                 Local<Object> ToObject(Isolate* isolate) const);
      |                               ^~~~~~~~
/usr/include/node/v8config.h:318:44: note: in definition of macro ‘V8_DEPRECATED’
  318 | #define V8_DEPRECATED(message, declarator) declarator
      |                                            ^~~~~~~~~~
/usr/include/node/v8.h:2590:31: note:   candidate expects 1 argument, 0 provided
 2590 |                 Local<Object> ToObject(Isolate* isolate) const);
      |                               ^~~~~~~~
```

Node.js 12.13.0

Propanu, 2019-11-11:
Hi @thiagomacieira this is due to SWIG not having Node.js 12 support released yet. There have been some developments on it though, they can be followed [here](https://github.com/swig/swig/issues/1520). Seems like there's a good chance this will get fixed with SWIG 4.1 since one of the project contributors has a working patch.

lissyx, 2019-11-12:
> Hi @thiagomacieira this is due to SWIG not having Node.js 12 support released yet. There have been some developments on it though, they can be followed [...]

Just to make it clear, I shared my changes in the hope that someone picks up on that and finished properly a PR. But I honestly don't have enough time to do that, not in the near future.