eclipse-upm/upm #694: RE: Compiling and UPM

silver2row, 2020-03-27:
Hello,

This seems to be an error on my side but I cannot be completely sure. 

    will change in GCC 7.1

I am compiling every module in python, python3, c, and c++ for now w/ make.

I am on an ARM board w/ a 32-bit processor, e.g. am335x from TI.com.

Seth

P.S. I looked through many of my errors during compilation. I am seeing a python 2.7.13 issue and the highlighted section in this post from above, gives me ideas that the GCC version is out of date on my board. Do you think I am on the right track so far to correct this issue?

silver2row, 2020-04-08:
Hello,

I know have this version of gcc:

    8.3.0

I am having severe issues w/ `make`.

See here:

make[2]: ***[src/2jciebu01_usb/CMakeFiles/jsupm_2jciebu01_usb.dir/build.make:63:src/2jciebu01_usb/CMakeFiles/jsupm_2jciebu01_usb.dir/CMakeFiles/jsupm_2jciebu01_usb.dir/2jciebu 01_usbJAVASCRIPT_wrap.cxx.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:869: src/2jciebu01_usb/CMakeFiles/jsupm_2jciebu01_usb.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

silver2row, 2020-05-02:
Hello,

I think I figured it out. Thank you for the repo. to list the issue.

Seth

closed 2020-05-02 by silver2row

tingleby, 2020-05-02:
@Propanu yeah lets turn of JS builds...

silver2row, 2020-05-15:
@tingleby ,

Also, JAVA for debian 10 is on the fritz from what I remember. I put in a bug report a while back but the people might have already been aware of this bug I spent time filling out. 

Just using default-jdk w/ apt install or default-jre gives errors on armhf boards on debian 10. 

Seth