eclipse-upm/upm #696: ImportError: cannot import name 'pyupm_jhd1313m1' from 'upm'

jpelletier, 2020-04-12:
When trying to run an example, it display this message.

babourn, 2020-05-18:
+1 Having the same issue with BMI160

babourn, 2020-05-18:
Partial fix/ possible issue with the install, I copied two python files into dist-packages in order to give python 3 access to the `*.py` files rather than the `*.so` files 
`sudo cp build/src/bmi160/python3.5/pyupm_bmi160.py /usr/local/lib/python3.5/dist-packages/upm`
`sudo cp build/interfaces/python3.5/pyupm_interfaces.py /usr/local/lib/python3.5/dist-packages/upm`

and that seemed to get everything working for my accelerometer.

jpelletier, 2021-02-01:
I tried babourn's fix (copied the python files from the upm library, cloned from github).
It worked for the LCD. There is a test program in humid_temp (examples from the Sensor mezzanine).

There is another issue preventing the communication between the dragonboard and the arduino.

g-vidal, 2021-02-01:
Hi @jpelletier 
I have not been on upm for long I did not take the time to read what happened and I might be out of the track, but I have compiled mraa and upm  to put them in a raspberryPi image I am distributing to colleagues.  I face the same problem  The reason is :
In folder `/usr/lib/pythonX.Y/dist-packages/upm` thereare only `_pyupm_*.so` libraries and no `pyupm_*.py` this could be normal if the `__init__.py` file contained all the necessary commands to retrieve module names from the libraries and unfortunately it is empty ! 
The workaround I found was to copy **all**`.py` files from the `src/pythonX.Y`  and `interfaces/pythonX.Y` folder . Your problem may appear  because of a dependency or interface problem.

I am using `jhd1313m1` `BME280` `TSL2561` `MCP9808` as a mini-weatherstation demo and everything is OK with upm.

here is the command I used in case it helps :
# from build folder
for i in python2.7 python3.7 ; do sudo cp src/*/$i/*.py /usr/lib/$i/dist-packages/upm/; done
for i in python2.7 python3.7 ; do sudo cp interfaces/$i/*.py /usr/lib/$i/dist-packages/upm/; done